JdeRobot/base #1: Compressed images in camera interface

rocapal, 2015-02-19:
Currently, the images are sent through camera interface using always a static size (320x240x3). It's interesting give the option to send the data (RGB or depth images) compressed.

For this, camera interface has to change and every components that use this interface.

hustcalm, 2015-03-22:
There is an implementation for compressing and uncompressing images and data in [RTAB-Map](http://introlab.github.io/rtabmap/), wonder if it does some help:

https://github.com/introlab/rtabmap/blob/master/corelib/src/Compression.cpp

rocapal, 2015-03-22:
Thanks for the pointer @hustcalm. RTAB-Map uses zlib in order to compress the images. Also we use zlib for compress the image structure of JdeRobot. Currently I'm working in the CIF branch (Compress Image Feature) and the results that I have are pretty good (see the image below).

[image]

hustcalm, 2015-03-22:
@rocapal 
Great job!!!
Glad to know it's working:-)

ojgarciab, 2015-03-29:
Looks like this commit doesn't work and Jenkins shows the same error: http://jenkins.jderobot.org/job/JdeRobot-GitHub/33/changes

Trying to compiling (new dependencies):

```
    -- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
    -- checking for one of the modules 'openssl'
    CMake Error at /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:367 (message):
      None of the required 'openssl' found
    Call Stack (most recent call first):
      Deps/openssl/CMakeLists.txt:3 (pkg_search_module)
      CMakeLists.txt:114 (include)
```

Fixed with:

```
    sudo apt-get install libssl-dev
```

So cmake:

```
    -- checking for one of the modules 'openssl'
    -- Using OpenSSL 1.0.1f
```

Doing a "make":

```
    Linking CXX executable cameraview
    /usr/bin/ld: ../../interfaces/cpp/jderobot/libJderobotInterfaces.so: referencia sin definir al símbolo 'pthread_mutexattr_settype@@GLIBC_2.2.5'
    /lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
    collect2: error: ld returned 1 exit status
    make[2]: *** [src/stable/components/cameraview/cameraview] Error 1
    make[1]: *** [src/stable/components/cameraview/CMakeFiles/cameraview.dir/all] Error 2
    make: *** [all] Error 2
```

Something around libJderobotInterfaces.so is break.

Regards,

ojgarciab, 2015-03-29:
PS: Refactor image interface leave a BIG problem: backward compatibility is completely broken.

You had not modified opencvdemo, but if I try to connect new opencvdemo version to my current AndroidCameraServer version, opencvdemo drops lots of messages about error protocol version.

If interfaces left untouched all applications are compatible until the last interfaces modification, so I think that is VERY important to change version number and document (in ChangeLog for example) to warning that anything compiled previously will be ever compatible until new recompile.

Another point is: Is it really necessary? **Compression could work fine without interface refactor. Server capabilities would be the same that client because both of them relies on colorspace functionality.**

It's simple: RGB8 still been the same RGB 8 bits per pixel without compression, RGB8z could be same RGB 8bpp but compressed. Compatibility: cameraserver (or androidcameraserver) could use or not compression with an command parameter (or #define), so it will send RGB8z format and colorspaces in client side will do the conversion to RGB8 if needed when execute:

```
    /* Get image with same format as origin */
    colorspaces::Image frame(image_data->description->width,
                             image_data->description->height, format_string,
                             &(image_data->pixelData[0]));
    /* Conversion to RGB8 will happen only if needed */
    colorspaces::ImageRGB8 frame_rgb8(frame);
```

I'm using a new format a few years ago (NV21) that was compatible with cameraview but not with opencvdemo because it only supports RGB8 because it didn't rely on colorspaces. When I refactorize opencvdemo I added that functionality and now I can use NV21 (and any other format supported by colorspaces).

**There is no need to create new methods or refactor interfaces.** Yet, maybe in a future version upgrade get image format and negotiate best will be useful, but right now it could be a fragmentation problem. Some components of jderobot 5.2 (or we are in 5.3) couldn't communicate with other components of jderobot 5.2.

IDEA: Attach changes in interfaces (not new ones, only changes in existing ones) to major version update.

I think that refactor Image would wait until jderobot 6.x version.

Conclusion: let colorspaces do it's job of converting from RGB8z to RGB8 if needed as done until today. The user could decide if server will use or not compression with a parameter (maybe) based in client capability (if it is a old one without compression or it supports compression) so until new minor (or major if decided) version the default will be uncompressed, but in new version the default will be compressed and a warning about it would be important.

Regards.

closed 2015-05-21 by jmplaza