JdeRobot/base #2: Multicamera calibration component

chanfr, 2015-02-19:
A new component JdeRobot is needed to calibrate many cameras at the same time. This component will calculate the intrinsic parameters at the same time as the extrinsic from a known pattern (chessboard), giving also the Rotation and Translation matrix between each pair of cameras.

hustcalm, 2015-03-22:
For two cameras, I know that `OpenCV` offers a routine:

```bash
A calibration example on stereo calibration can be found at opencv_source_code/samples/cpp/stereo_calib.cpp
```

For many(more than two), maybe we can design the module based on stereo calibration? I don't know, just a little thought:-)

chanfr, 2015-03-22:
Hi,
the idea is to integrate a calibration method into the JdeRobot project. The main idea of this project is not to estimate only the intrinsic parameters but also the extrinsics. OpenCv gives us some of the required routines and we want to put them all together to get that functionality. The following scheme shows the idea of the component where we will get capture images from jderobot standard camera interface and the output is the intrinsics of each camera and also the RT between each camera and the pattern.

[image]

 With this approach we get also the relationship between all the calibrated cameras using the appropriate RT combination. This relationship can be done as we will only use synchronized frames where the pattern has been detected in all cameras.

This component is already implemented I'm just testing it and will be updated in a few days.

hustcalm, 2015-03-23:
@chanfr 
Thanks for the clear explanation and illustration, I got the idea finally:-)

closed 2015-03-31 by chanfr