JdeRobot/base #5: Error when running `cmake .`, no python files

hustcalm, 2015-03-17:
Hi there,

Trying to build `JdeRobot` following the instructions [here](https://github.com/RoboticsURJC/JdeRobot), however when running `cmake .`, I got the following errors:
```bash
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/body_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/exceptions_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/common_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/bodyencoders_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/motors_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/replayControl_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/namingService_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/varcolor_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/remoteConfig_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/bodymovements_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/kinectleds_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/datetime_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/sonars_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/ptmotors_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/remoteCloud_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/naofollowball_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/pose3d_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/camera_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/pointcloud_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/containers_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/image_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/recorder_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/pose3dencoders_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/wiimote_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/jcm_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/encoders_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/bodymotors_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/naomotions_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/laser_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/pose3dmotors_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/jointmotor_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/genericData_ice.py': No such file or directory
/usr/bin/slice2py: error: cannot open`/home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/ptencoders_ice.py': No such file or directory
```

However, these issues don't bother too much, since `cmake` still succeeds and generates the `Makefile`.

Maybe we need to clean the `CMakeLists.txt` file a bit?

Regards,
Lihang

rocapal, 2015-03-17:
Hi!

Please, just create the directory as following:
```bash
mkdir -p /home/lihang/dev/gsoc2015/JdeRobot/src/stable/interfaces/slice/jderobot/../../python/jderobot/
```
And then type cmake again. We are trying to fix it in the repository.

hustcalm, 2015-03-18:
@rocapal 

Cool, it works! 

Should we create the directory using the CMake scripts or fix this by modifying the source code? I'd like to do some hack and fix this:-)

Thanks,
Lihang

rocapal, 2015-03-18:
@hustcalm I think it's better add some lines in CMakeLists.txt file to create the directory. Empty directories in the repository structure is not clean.

You can add the lines that you need after macros in CMakeLists.txt file.

Thanks!

hustcalm, 2015-03-18:
@rocapal I have pulled a pull request #7, please kindly have a review.

Thanks,
Lihang

chanfr, 2015-03-18:
Your pull request have been tested and approved.

thanks!

commit cee710e8 (2015-03-18): Merge pull request #7 from hustcalm/fix_python_path

Fixed #5 by making the python/jderobot directory using CMake macros
commit d129c085 (2015-03-18): Fixed #5 by making the python/jderobot directory using CMake macros
closed 2015-03-18 by chanfr