JdeRobot/base #12: Refactor all the source code comments to English

hustcalm, 2015-03-22:
I also saw there is a discussion in the mailing list about language. As I'm reading the source code of JdeRobot, I find there are some comments maybe in Spanish which can be confused to other developers.

There are a few lines of comments in the root `CMakeLists.txt` and I don't for sure if there are any in other source code files either.

Though this is not a big issue, using English might make other developers feel easier to get onboard:-)

rocapal, 2015-03-22:
You are right @hustcalm. We are in the process to internationalize the project and we need to be carefuly with this things. We will use this issue to manage this translations

Thanks a lot for your comments!

hustcalm, 2015-03-22:
@rocapal 
Cool, so glad to hear that you are already on the way!!!
I got another idea, should the mailing list be sent in English?
I'm wondering if I miss something, I receive the mailing list digest in Spanish maybe? Though the topics are discussed in English now. Maybe we should make it universal and all English?

abhinavjain241, 2015-03-22:
@rocapal 

This would be a nice idea, and it'll be easier for developers from all across the world to make changes to the source code. I would love to contribute here!

RoboticsURJC, 2015-03-22:
Hi @hustcalm <https://github.com/hustcalm>,

we actually changed the official language for the list messages to English
some days ago. In addition I've just changed the configuration of mailman
server, from now on you should receive the list digests in English too,
both the headers and the contents. Please let us know whether is working
properly for you.

JoseMaria

hustcalm, 2015-03-23:
Hi @RoboticsURJC ,

Thanks for the update, I make it right for now. In case of others who confused about this, here are my steps for making it right:-)

1. Visit [JdeRobot Mailman](http://gsyc.escet.urjc.es/cgi-bin/mailman/listinfo/jde-developers) and ensure the display language is English
[image]

2. Log in by inputing you subscriber Email
[image]

3. Scroll and find the options below
[image]

<hr />

It' easier for new subscribers, just make sure you select:
[image]

Cheers,
Lihang

gsssrao, 2015-03-26:
This would be very helpful for the community and other users. I am interested in contributing here!

closed 2021-02-21 by jmplaza