JdeRobot/base #13: Describe the process to contribute in JdeRobot project

rocapal, 2015-03-22:
We need to describe the process and policy to third-people in order to contribute in JdeRobot project.

It would be nice have a description and screenshots of the next steps:

* Create a fork
* Modify source code
* Pull request

Here there is a guide of github that we can use: https://guides.github.com/activities/contributing-to-open-source/

hustcalm, 2015-03-22:
@rocapal 
I'd like to try and hopefully will get this done for one or two days.
So you can assign the issue to me:-)

hustcalm, 2015-03-23:
@rocapal 
I was just trying to edit the [wiki page](https://github.com/RoboticsURJC/JdeRobot/wiki/) to place the contribution process there. However, I find it better to put a draft here maybe. The wiki pages can be filled later on. Here it goes:-)

<hr />

It's very cool to use Git and Github as the collaborate tools for JdeRobot development. 

The basic workflow would be like:

* Fork the JdeRobot repository
* Clone your own JdeRobot repository using Git
* Create a new branch for new features and bug fixes
* Work on your branch and get it run and tested
* Commit your changes and open a pull request
* Make changes regarding the discussions of your pull request

For now, I'm gonna demo this workflow using my Github account and try to add a dummy feature, since the contribution process is not so well documented, I will start here to add a section named `How To Contribute` to the `README.md` file of the project. Here we go!!!

## Fork the JdeRobot repository
First of all, to make your changes accepted as a contributor, we have to `Fork` the repository. For more roles of a Github project, see [here](https://guides.github.com/activities/contributing-to-open-source/#project).

To fork a repository on Github, just click on the right corner of the project page as illustrated below:
[image]

Then follow the panel to choose where your fork should go, typically you will select your personal Github account if you got many choices, for me I choose(@hustcalm) as:
[image]

After that, you should be redirected to the new repository under your account, for me, it looks like:
[image]

## Clone your own JdeRobot repository using Git
Once you have forked the repository, it's time to start some hacks on your repository! However, first of all, you have to clone the source code using `Git`:
[image]

Basically, you have 3 options:

* HTTPS
* SSH
* Subversion

No big differences, just different protocols. If you have set up your [Github SSH](https://help.github.com/articles/generating-ssh-keys/), I suggest using `SSH`. For using `HTTPS`, you will be prompted to input user name and password when accessing your `Github` repository from terminal. 

Now, open your terminal and `cd` to a proper directory, then `clone`:
[image]

## Create a new branch for new features and bug fixes
For contribution, developers will always add new features or fix bugs. For easy development management, we will use `branches`. For default branches after a fresh clone, you can see them by `git branch -a`:
[image]

To create a new branch, use `git branch branch_name`, this will create a new branch based on your current working branch. Like `git branch JdeRobot_how_to_contribute`:
[image]

Switch to the new branch using `git checkout branch_name`, this will make `branch_name` the current working branch. Like `git checkout JdeRobot_how_to_contribute`:
[image]

OK, let's have a check to just make sure, you can always run `git status` or `git remote` to see what is going on with your local repository:
[image]

For more useful commands about `Git`, you can always run `git help`.

## Work on your branch and get it run and tested
Well, really time to hack! Now you can concentrate on your new feature or bug fix and show your talent:-)

As for the demo, I will edit the `README.md` file, like:
[image]

The edited file would be like:
[image]

After your hack, make sure your changes are doing well, especially not destroying the compiled source code. Tests are great ways to make things right, so please test your changes and make sure everything is just fine!

## Commit your changes and open a pull request
Once you are done with your feature or bug fix, time to share with the community:-)

Basically, you should:

### Commit your changes
[image]

### Push to your repository
[image]

### Create a pull request
[image]

To create a good pull request, be sure to make things clear:

[image]

## Make changes regarding the discussions of your pull request
[image]

Cheers, any comments are welcome, let's make it better together:-)
Lihang

hustcalm, 2015-03-23:
Created a new wiki page [here](https://github.com/RoboticsURJC/JdeRobot/wiki/How-To-Contribute), hope it helps:-)

rocapal, 2015-03-23:
@hustcalm awesome!! It's perfect. I'll include it in the wiki of JdeRobot, but just I link the wiki page of github that you have created.

Thanks a lot!!

hustcalm, 2015-03-24:
@rocapal 
I'm glad it helps and feels really good contributing to JdeRobot:-)

I talked with JoseMaria earlier by Email and maybe I can have write access on the official wiki pages. So I'm thinking about maintaining the `How To Contribution` both on `Github` and `jderobot.org`.

Any feedback is welcome, I'm always ready to polish it:-)

Here is a good [How To Contribution sample](https://wiki.openstack.org/wiki/How_To_Contribute) coming from `OpenStack` project, I think we can do more!!!

Cheers,
Lihang

rocapal, 2015-03-26:
Closed! If we need some documentation about contribute in the project, we should use this issue again.

commit 86560427 (2015-03-29): Merge pull request #15 from hustcalm/JdeRobot_doc_How_To_Contribute

Fix #13 - Add a How To Contribute section
commit 928bd4e0 (2015-03-29): Fix #13 - Add a How To Contribute section
closed 2015-03-26 by rocapal