JdeRobot/base #17: Warning about sscanf string format

ojgarciab, 2015-03-24:
There is a warning while compile that is about sscanf string format:

```
Scanning dependencies of target rgbdManualCalibrator
[ 91%] Building CXX object src/stable/components/rgbdManualCalibrator/CMakeFiles/rgbdManualCalibrator.dir/controller.cpp.o
[ 92%] Building CXX object src/stable/components/rgbdManualCalibrator/CMakeFiles/rgbdManualCalibrator.dir/drawarea.cpp.o
/home/redstar/Documentos/jderobot/temp/JdeRobot/src/stable/components/rgbdManualCalibrator/drawarea.cpp: In member function ‘int rgbdManualCalibrator::DrawArea::load_line(FILE*)’:
/home/redstar/Documentos/jderobot/temp/JdeRobot/src/stable/components/rgbdManualCalibrator/drawarea.cpp:520:162: warning: format ‘%s’ expects a matching ‘char*’ argument [-Wformat=]
   number=sscanf(buffer_file,"%s %s %s %s %s %s %s %s %s %s %s %s",word1,words[0],words[1],words[2],words[3],words[4],words[5],words[6],words[7],words[8],words[9]);
                                                                                                                                                                  ^
[ 92%] Building CXX object src/stable/components/rgbdManualCalibrator/CMakeFiles/rgbdManualCalibrator.dir/rgbdManualCalibrator.cpp.o
[ 93%] Building CXX object src/stable/components/rgbdManualCalibrator/CMakeFiles/rgbdManualCalibrator.dir/view.cpp.o
Linking CXX executable rgbdManualCalibrator
[ 93%] Built target rgbdManualCalibrator
Scanning dependencies of target rgbdViewer
[ 93%] Building CXX object src/stable/components/rgbdViewer/CMakeFiles/rgbdViewer.dir/rgbdViewer.cpp.o
[ 93%] Building CXX object src/stable/components/rgbdViewer/CMakeFiles/rgbdViewer.dir/drawarea.cpp.o
/home/redstar/Documentos/jderobot/temp/JdeRobot/src/stable/components/rgbdViewer/drawarea.cpp: In member function ‘int rgbdViewer::DrawArea::load_line(FILE*)’:
/home/redstar/Documentos/jderobot/temp/JdeRobot/src/stable/components/rgbdViewer/drawarea.cpp:524:162: warning: format ‘%s’ expects a matching ‘char*’ argument [-Wformat=]
   number=sscanf(buffer_file,"%s %s %s %s %s %s %s %s %s %s %s %s",word1,words[0],words[1],words[2],words[3],words[4],words[5],words[6],words[7],words[8],words[9]);
                                                                                                                                                                  ^
[ 94%] Building CXX object src/stable/components/rgbdViewer/CMakeFiles/rgbdViewer.dir/rgbdViewergui.cpp.o
[ 94%] Building CXX object src/stable/components/rgbdViewer/CMakeFiles/rgbdViewer.dir/myprogeo.cpp.o
```

I do not know why this issue has not been fixed so far. It makes me think that the reason is to avoid another problem.

If this code was wrote this way to avoid some strange behavior when the line parsed has more data than expected then I think that is better to do this in files drawarea.cpp from rgbdViewer and rgbdManualCalibrator components:

```
@@ -477,7 +477,7 @@
 DrawArea::load_line(FILE *myfile)
 {
 	const int limit = 256;	
-	char word1[limit],words[10][limit];
+	char word1[limit],words[11][limit];
 	int i=0;
 	char buffer_file[limit];  
 	int number;
@@ -517,7 +517,7 @@
 		return 0; 
 	/* return EOF; empty line*/
 	else {
-		number=sscanf(buffer_file,"%s %s %s %s %s %s %s %s %s %s %s %s",word1,words[0],words[1],words[2],words[3],words[4],words[5],words[6],words[7],words[8],words[9]);
+		number=sscanf(buffer_file,"%s %s %s %s %s %s %s %s %s %s %s %s",word1,words[0],words[1],words[2],words[3],words[4],words[5],words[6],words[7],words[8],words[9],words[10]);
 		if (strcmp(word1,"worldline")==0){
 			if (number == 9){
 				if (numlines <MAX_LINES){
```

So there is a extra space for data (if it exists) or do this if this is not the problem:

```
@@ -517,7 +517,7 @@
 		return 0; 
 	/* return EOF; empty line*/
 	else {
-		number=sscanf(buffer_file,"%s %s %s %s %s %s %s %s %s %s %s %s",word1,words[0],words[1],words[2],words[3],words[4],words[5],words[6],words[7],words[8],words[9]);
+		number=sscanf(buffer_file,"%s %s %s %s %s %s %s %s %s %s %s",word1,words[0],words[1],words[2],words[3],words[4],words[5],words[6],words[7],words[8],words[9]);
 		if (strcmp(word1,"worldline")==0){
 			if (number == 9){
 				if (numlines <MAX_LINES){
```

To get the exact number of string as parameters.

Could someone test this patches to decide between one solution or another?

Thanks.

hustcalm, 2015-03-25:
@ojgarciab 
I think the solution should depend on the input `buffer_file`. 

However according to the [sscanf](http://www.cplusplus.com/reference/cstdio/sscanf/) and [scanf](http://www.cplusplus.com/reference/cstdio/scanf/) manual,

> Depending on the format string, the function may expect a sequence of additional arguments, each containing a pointer to allocated storage where the interpretation of the extracted characters is stored with the appropriate type.
There should be at least as many of these arguments as the number of values stored by the format specifiers. Additional arguments are ignored by the function.

Maybe @chanfr and @rocapal can help with this:-)

ojgarciab, 2015-03-25:
Hi Lihang, thanks for answering.

I'm not talking about C syntax, I am talking about that the solution is about the reason to add an extra %s.

The sscanf format text have 12 %s but there is 11 format parameters.

If it was a typo mistake then we could remove it safely, but if it was a workaround to fix a issue about malformed configuration files then we need to add an extra parameter.

Last problem is not

hustcalm, 2015-03-25:
@ojgarciab 
Sure, maybe I don't make myself clear enough. Agreed that the original developer should resolve this issue.

commit ace1b975 (2015-03-31): Merge pull request #21 from ojgarciab/Fix_17

Fix #17: Warning about sscanf string format in two files drawarea.cpp
commit c3f945b7 (2015-03-31): Fix #17: Warning about sscanf string format in two files drawarea.cpp
closed 2015-03-31 by chanfr