JdeRobot/base #19: Proposal for a .gitignore file

ojgarciab, 2015-03-25:
To prevent tens of untracked files to be shown during every commit it's a good idea to use a .gitignore file.

I propose a minimal pattern (*.a, *.so, Makefile, ect) and if you wish we could add every component's executable name.

But I think that at least a minimal filtering must be implemented to simplify commit and status output.

hustcalm, 2015-03-25:
@ojgarciab 
Great, I'm also thinking about that a `.gitignore` file should help a lot:-)

Maybe we should also write a good pattern to all the generated files, like:
```bash
	src/stable/components/basic_component/basic_component
	src/stable/components/basic_component_qt/basic_component_qt
	src/stable/components/basic_component_qt/gui/moc_gui.cxx
	src/stable/components/basic_component_qt/gui/moc_gui.cxx_parameters
	src/stable/components/basic_server/basic_server
	src/stable/components/basic_server/myInterface.cpp
	src/stable/components/basic_server/myInterface.h
	src/stable/components/cameraserver/cameraserver
	src/stable/components/cameraview/cameraview
	src/stable/components/cameraview_icestorm/cameraview_icestorm
	src/stable/components/colorTuner/colorTuner
	src/stable/components/followBall/followball
	src/stable/components/giraffeClient/giraffeClient
	src/stable/components/giraffeServer/giraffeServer
	src/stable/components/introrob/introrob
	src/stable/components/introrob_qt/gui/moc_gui.cxx
	src/stable/components/introrob_qt/gui/moc_gui.cxx_parameters
	src/stable/components/introrob_qt/gui/moc_stategui.cxx
	src/stable/components/introrob_qt/gui/moc_stategui.cxx_parameters
	src/stable/components/introrob_qt/gui/moc_threadupdategui.cxx
	src/stable/components/introrob_qt/gui/moc_threadupdategui.cxx_parameters
	src/stable/components/introrob_qt/gui/widget/moc_cameraswidget.cxx
	src/stable/components/introrob_qt/gui/widget/moc_cameraswidget.cxx_parameters
	src/stable/components/introrob_qt/gui/widget/moc_controlvw.cxx
	src/stable/components/introrob_qt/gui/widget/moc_controlvw.cxx_parameters
	src/stable/components/introrob_qt/gui/widget/moc_glwidget.cxx
	src/stable/components/introrob_qt/gui/widget/moc_glwidget.cxx_parameters
	src/stable/components/introrob_qt/gui/widget/moc_laserwidget.cxx
	src/stable/components/introrob_qt/gui/widget/moc_laserwidget.cxx_parameters
	src/stable/components/introrob_qt/introrob_qt
	src/stable/components/introrob_qt/moc_depuratewindow.cxx
	src/stable/components/introrob_qt/moc_depuratewindow.cxx_parameters
	src/stable/components/introrob_qt/pioneer/moc_pioneer.cxx
	src/stable/components/introrob_qt/pioneer/moc_pioneer.cxx_parameters
	src/stable/components/introrob_qt/pioneer/moc_pioneeropengl.cxx
	src/stable/components/introrob_qt/pioneer/moc_pioneeropengl.cxx_parameters
	src/stable/components/introrob_qt/qrc_resources.cxx
	src/stable/components/introrob_qt/resources.qrc.depends
	src/stable/components/introrob_qt/robot/moc_actuators.cxx
	src/stable/components/introrob_qt/robot/moc_actuators.cxx_parameters
	src/stable/components/introrob_qt/robot/moc_robot.cxx
	src/stable/components/introrob_qt/robot/moc_robot.cxx_parameters
	src/stable/components/introrob_qt/robot/moc_sensors.cxx
	src/stable/components/introrob_qt/robot/moc_sensors.cxx_parameters
	src/stable/components/introrob_qt/robot/moc_threadupdaterobot.cxx
	src/stable/components/introrob_qt/robot/moc_threadupdaterobot.cxx_parameters
	src/stable/components/namingService/namingService
	src/stable/components/opencvdemo/opencvdemo
	src/stable/components/recorder/recorder
	src/stable/components/remoteConfiguration/remoteConfiguration
	src/stable/components/replayController/replayController
	src/stable/components/replayer/replayer
	src/stable/components/rgbdCalibrator/rgbdCalibrator
	src/stable/components/rgbdManualCalibrator/rgbdManualCalibrator
	src/stable/components/rgbdViewer/rgbdViewer
	src/stable/components/teleoperator/teleoperatorPC
	src/stable/interfaces/python/
```
Instead of writing every generated file's name to `.gitignore`, any better idea?

ojgarciab, 2015-03-25:
Hi Lihang, thanks for your idea.

I think that proposed pattern covers almost all of those extra generated files:

```
CMakeCache.txt
CMakeFiles/
*.cmake
*.a
*.so
*.cxx
*.cxx_parameters
lib*.so.*
Makefile
```

After a clean clone, before cmake and make, there is those files without matching proposed pattern (qt related components are not compiled so cxx files are not generated):

```
	src/stable/components/basic_component/basic_component
	src/stable/components/basic_server/basic_server
	src/stable/components/basic_server/myInterface.cpp
	src/stable/components/basic_server/myInterface.h
	src/stable/components/cameraserver/cameraserver
	src/stable/components/cameraview/cameraview
	src/stable/components/cameraview_icestorm/cameraview_icestorm
	src/stable/components/colorTuner/colorTuner
	src/stable/components/followBall/followball
	src/stable/components/giraffeClient/giraffeClient
	src/stable/components/introrob/introrob
	src/stable/components/namingService/namingService
	src/stable/components/naoviewer/naooperator
	src/stable/components/opencvdemo/opencvdemo
	src/stable/components/recorder/recorder
	src/stable/components/remoteConfiguration/remoteConfiguration
	src/stable/components/replayController/replayController
	src/stable/components/replayer/replayer
	src/stable/components/rgbdCalibrator/rgbdCalibrator
	src/stable/components/rgbdManualCalibrator/rgbdManualCalibrator
	src/stable/components/rgbdViewer/rgbdViewer
	src/stable/components/teleoperator/teleoperatorPC
	src/stable/components/visualHFSM/visualHFSM
	src/stable/interfaces/cpp/
	src/stable/interfaces/python/
```

Maybe it's a good idea to ignore executable names, but I think that it is more important ignore interfaces directory (cpp and python) to avoid uploading them for a mistake.

I'll add your contribution (*.cxx and *.cxx_parameters) and interface directories.

Thanks.

commit 32fe2299 (2015-03-25): Fix #19: New pattern for qt generated files and interface directories
commit 63318882 (2015-03-25): Merge pull request #20 from ojgarciab/Fix_19

Fix #19: Proposal for a minimal .gitignore file
commit ab51f74b (2015-03-25): Fix #19: Proposal for a minimal .gitignore file
closed 2015-03-25 by rocapal