JdeRobot/base #27: Reformat opencvdemo component

ojgarciab, 2015-03-25:
I would like to fix, as an exercise/practice, this:

- Convert real tabs in 4 space ones.
- Correct syntax as proposed.
- Remove `using namespace cv;` (it is a bad practice). Add cv:: prefix to all OpenCV functions/clases.
- Fix old naming conventions (C styled OpenCV, as `cvLine` -> `cv::line`) or deleting old comments holding unused code.

Best regards.

ojgarciab, 2015-03-25:
First problem found: [The #define Guard](https://google-styleguide.googlecode.com/svn/trunk/cppguide.html#The__define_Guard)

>All header files should have #define guards to prevent multiple inclusion. The format of the symbol name should be <PROJECT>_<PATH>_<FILE>_H_.
>
>To guarantee uniqueness, they should be based on the full path in a project's source tree. For example, the file foo/src/bar/baz.h in project foo should have the following guard:
>
>     #ifndef FOO_BAR_BAZ_H_
>     #define FOO_BAR_BAZ_H_
>     
>     ...
>     
>     #endif  // FOO_BAR_BAZ_H_

So `src/stable/components/opencvdemo/viewer.h` header file should be:

```
#ifndef JDEROBOT_STABLE_COMPONENTS_OPENCVDEMO_VIEWER_H_
#define JDEROBOT_STABLE_COMPONENTS_OPENCVDEMO_VIEWER_H_

...

#endif  // JDEROBOT_STABLE_COMPONENTS_OPENCVDEMO_VIEWER_H_
```

It looks so large... but for now I'll use that guard format.

abhinavjain241, 2015-03-25:
I think I can work on fixing this issue. I'll send a pull request in 1-2 days once I'm done re-factoring the source code.

ojgarciab, 2015-03-25:
I'm working on it right now. It's almost finish.

abhinavjain241, 2015-03-26:
Oh okay okay, thats fine. I'd love to help by suggesting enhancements :)

ojgarciab, 2015-03-26:
Your enhancements are welcome.

I just have done some code optimizations to immunize it about common network errors (time out when remote server are offline and connection refused when remote server is online but ice endpoint is not running).

When I finish this refactorize you could add your own code to enhance it :)

ojgarciab, 2015-03-27:
I added support to every image format supported by `colorspaces::Image`, included NV21 (for android devices). Before that update only `RGB8` format was supported by `opencvdemo`.

aitormf, 2017-05-12:
Can I close this issue @ojgarciab ??

ojgarciab, 2017-08-31:
Your can close it, it is too old and it's too late to merge changes.

closed 2017-08-31 by ojgarciab