JdeRobot/base #28: Jenkins for Continous Integration

hustcalm, 2015-03-26:
Hi there,

It just came to my mind, wondering if we need a Jenkins CI to ensure the smooth of development process. It's important that the new pushes compile and do no harm to the code base.

Any idea @rocapal @ojgarciab ?

Github offers good services for getting Jenkins on board, see [1][2].

[1] https://gist.github.com/tehfoo/3708738
[2] http://cvuorinen.net/2013/06/installing-jenkins-ci-server-with-github-integration-for-a-php-project/

ojgarciab, 2015-03-26:
We have Jenkins working here: http://jenkins.jderobot.org/job/JdeRobot-GitHub/

@rocapal is the main maintainer of Jenkins platform.

hustcalm, 2015-03-26:
Thanks for the hint @ojgarciab.

Great then, for better integration with Github, maybe we can try [Travis CI](https://travis-ci.org/recent) @rocapal.

Just a thought, no hurry:-)

rocapal, 2015-03-26:
@hustcalm Github has some hooks for trigger the Jenkins compilation. Right now, for each push in the repository a jenkins petition to compile is sent.

Are you thinking in other kind of integration? Also github has webhooks for Travis CI

hustcalm, 2015-03-26:
@rocapal I see that there will be a small image like:
[image]

when using `Travis CI`, I don't know if using `Jenkins` can also do this. It would be nice if we can see the status of the building on the index page of the `Github repository` rather than visiting the specified `Jenkins URL`.

What do you think then?

rocapal, 2015-03-26:
Yes, we have the same in Jenkins. Just check it: http://jenkins.jderobot.org/buildStatus/icon?job=JdeRobot-GitHub

Anyway, if compilation process fails jenkins send an email to the last commiter and the JdeRobot's admin.

I leave open this issue until we integrate this link in README file, for instance.

hustcalm, 2015-03-26:
Cool!!!

I just added the link in README file, see #29.

closed 2015-03-26 by hustcalm