JdeRobot/base #31: Creating deb packages for the new release of JdeRobot

fqez, 2015-03-26:
Hi, I'm opening this issue to discuss all the changes related with the new release of JdeRobot (v5.3) in which I'm working right now. The idea of the new release is to upate all our third party softwares to their last versions, for instance:

* Gazebo 1.8.1 --> Gazebo 5.1
* OpenCV 2.3.1 --> OpenCV 2.4
* Ice 3.4 --> Ice 3.5
* etc

modifying the components affected by the update. The goal is to make a deb package easy to install from our repository with the following command:

`sudo apt-get install jderobot` and `sudo apt-get install jderobot-dev` (to download the headers, etc).

I've been doing some changes to some CMakeLists files into the project in order to make the new versions compatible with JdeRobot, but due to we are un a hurry now because of the Drones contest that is coming (next month) the changes are not clean and elegant, so it's certain that we will need to look over them after the contest. But from now, I've managed to create a package with CPack, but I have a little problem when I try to install it with `dpkg -i`:

```
$ sudo dpkg -i jderobot_5.2.4-rc1_i386.deb 
[sudo] password for fran: 
(Reading database ... 246408 files and directories currently installed.)
Preparing to unpack jderobot_5.2.4-rc1_i386.deb ...
Unpacking jderobot (5.2.4-rc1) over (5.2.4-rc1) ...
dpkg: dependency problems prevent configuration of jderobot:
 jderobot depends on openni2; however:
  Package openni2 is not installed.

dpkg: error processing package jderobot (--install):
 dependency problems - leaving unconfigured
Processing triggers for libc-bin (2.19-0ubuntu6.6) ...
Errors were encountered while processing:
 jderobot
```

If I install JdeRobot from source `cmake .` and `make` work fine and find OpenNi2, but if I try to install JdeRobot from the package I've already created it shows the error above :disappointed: 
I've modified the CMakeLists of openni2 folder adding the following line (to make it compile) `SET(OPENNI2_LIB_PATH /usr/lib)` for your information. Any ideas?

rocapal, 2015-03-26:
@fqez the package dependencies are resolved when the package is installed from a repository. If you install single package using 'dpkg -i' there is not any process to resolve dependencies. Just it happens if you use 'aptitude install $package'.

To become familiar with this process I suggest to the next steps (I can help you):
* 1. First, we need to build the chroot of ubuntu 14.04
* 2. Compile evert thrid-part software that it's not in the ubuntu repository (like gazebo, pcl, openniserver, etc).
* 3. Create the debian package and upload this debian packages to JdeRobot debian repository
* 4. Install JdeRobot successfully :-)

So, let's start with the first step. What do you think?

fqez, 2015-03-26:
@rocapal yes yes I knew that `dpkg -i` doesn't resolve the dependencies, but report them! The fact is that all the other dependencies are fine, but the one that is giving me problems is openni2. It does not find it although openni2 is installed (cause if I compile from source it works fine). 
Thank you for the steps, now is more clear to follow a path to archieve this issue!
About the first step, I use debootstrap and schroot to make a chroot so I think I can do it with no problems (If they come up I will put a message here).
The second step is already completed in a virtual machine with Ubuntu 14.04 i386 installed in it (I have to test with amd64), I will repeat the process taking notes of every step in a chrooted environment.
The third step is also completed (If i did it right) but I have to test it.
About uploading it to the JdeRobot repository I have no clue about how to do that at all, but I will ask for help when I get that step. 

So yes, let's start from the beggining, I will post the instructions to create a chroot of Ubuntu 14.04 amd64.

Thanks a lot! :smile:

rocapal, 2015-03-26:
@fqez openni2 package is in our repository[1], but it's not updated for ubuntu 14.04 so don't try to install from it.

To documentation the process, please use the wiki of github. Later we'll decide where finally goes the content.

[1] http://jderobot.org/Downloads#Debian_packages_from_our_APT_Repository

fqez, 2015-03-26:
@rocapal yes, I know. I installed openni2 from PCL repositories. Ok I will use the wiki.
Thank you.

fqez, 2015-04-01:
Hi, I've added a new page to the wiki explaining the process to create a chrooted environment. Also, I've created a chroot for:

* Ubuntu 14.04 Trusty amd64 and i386
* Debian Wheezy (stable) amd64 and i386

Just need to be created:

* Debian Jessie (testing) amd64 and i386

Once created, I will upload a new page to the wiki describing how to generate debian packages of JdeRobot using CPack.

Best,

abhi-kumar, 2015-04-01:
Why are we just upgrading from OpenCV 2.3 to 2.4 and not to 3.0, 3.0 version being pretty stable and providing with newer modules like advanced image stitchers, blenders, object recognition modules and trackers.

rocapal, 2015-04-01:
@fqez  perfect! Good work! I'have seen the howto to create the chroots ant it's perfect. At the moment is enough with Ubuntu 14.04 (amd64/i386) and Debian wheezy (amd64/i386). I will not create debian packages for debian testing (at the moment).

@abhi-kumar opencv 3.0 is still beta [1]. We can continue this discussion in your push request [2]

[1] http://opencv.org/downloads.html
[2] https://github.com/RoboticsURJC/JdeRobot/pull/40

fqez, 2015-04-03:
Hi @rocapal 
I'm trying to create a deb package for Debian stable (wheezy) amd64, but I can't find out how to install openni2. The official repository of PCL does not install openni2 in Debian (it does in Ubuntu). Should I generate a package of OpenNi2 for debian from source?? Or there is a more simple way to install it.
Thanks!

fqez, 2015-04-03:
Also I've just read that Openni2 is not supported by Debian Wheezy [here](http://wiki.icub.org/wiki/OpenNI2#Installation). It's that true?

abhi-kumar, 2015-04-03:
@fqez  here's a repository supporting debian-openni2 package : https://github.com/atuleu/debian-openni2. Also openni2 has some issues with PCL : http://www.pcl-developers.org/OpenNI2-td5707600.html

fqez, 2015-04-03:
Hi @abhi-kumar, first of all thank you so much for the reply. I found that repository before and I did the `make`, but when I do `cmake .` to compile JdeRobot it seems to not find libopenni2... here is the output:

```
-- checking for module 'libopenni2'
--   package 'libopenni2' not found
-- Could NOT find OpenNI2 (missing:  OPENNI2_INCLUDE_DIRS) 
```
Did you manage to install it properly? Thank you again!

abhi-kumar, 2015-04-03:
@fqez I will try to compile the repository and let you know accordingly in some time.

abhi-kumar, 2015-04-03:
@fqez I compiled the source files, it  is a modification of the repository at https://github.com/OpenNI/OpenNI2 (OpenNI2-master). Currently the development is going on at this occipital repo: https://github.com/occipital/openni2 , where it is mentioned as to how to create a final package in the usr/include as jdeRobot tracks openni2 only from usr/include directory

fqez, 2015-04-07:
Hi @rocapal I've managed to create a package for Ubuntu 14.04 amd64 and I'd like to upload it to the JdeRobot debian repository (step 3) to test if it works properly and installs everything well. How should I procceed?

rocapal, 2015-04-07:
@fqez regarding openni2 package. We have the sources in JdeRobot repository to create the openni2 package. In this way we use always a stable version. I'm going to try put the sources that we have in a public GIT repository 'JdeRobot-ThirdParty'. This involves more work of us in order to update with new versiones. But the good part is that JdeRobot platform is very stable.

For the debian repository, we need to create it (The repository that we have is older and we need to update it). You can test in 'ordago' machine and when it's ready we can migrate this debian repository to jderobot.org, what do you think? The debian repository should contain the sources as well, to allow download it using "apt-get source" command.

rocapal, 2015-04-07:
@fqez I have already uploaded our lastest sources of openni2. You can find in the new git repository https://github.com/RoboticsURJC/JdeRobot-ThirdParty

To create debian packages of openni2 just type:
```
$ dpkg-builpackage -us -uc -rfakeroot
```

Then, the following files are generated:
```
openni2_2.2.0-2jderobot_amd64.deb  
openni2_2.2.0-2jderobot_amd64.changes
openni2_2.2.0-2jderobot.dsc  
openni2_2.2.0-2jderobot.tar.gz
```
The first one is the debian package, and the rest of the files are needed to build the source repository.

Try to build the packages and let me know if you have any problem.

BTW, we need to write documentation about which third software we need to JdeRobot platform, or which apt repository we need (for instance, PCL). We need to create a list of every software we need and they are not in the debian/ubuntu official repositories.

fqez, 2015-04-07:
Hi!
I've generated a .deb of JdeRobot using Cpack from our last source version on github in a jail in 'ordago'!. About openni2, in Ububtu 14.04 it can be installed from the pcl repository (at least it was installed from it yesterday) but I'm not sure if it is the lastest version. I'm also going to create a package of openni2 as you said, no problem.
My main doubt is about creating a repository... due to I have no experience with that. Nevertheless I can do a little search in Google and give it a try! There will be no problem about creating a whole new repository.
Yes, don't worry about the documentation, I am saving all the information about this issue :)

fqez, 2015-04-07:
@rocapal The generation of Openni2 package works perfectly!!

fqez, 2015-04-12:
Hi,

I've been searching a lot of information about creating a debian repository and I found a easy solution to do that. However I don't know if it would be valid for us. The process is explained [here](http://enavas.blogspot.com.es/2012/10/crear-nuestro-propio-repositorio-de.html)
What do you think @rocapal?
We can put the openni2 package and the JdeRobot package in there.

Regarding to Debian, I've found that some versions differ from Ubuntu ones. for instance, Gazebo is only available for its 3 version( v5 in Ubuntu), Boost for its 1.55 version (v1.54 in Ubuntu) and PCL does not install properly from official packages, while Ubuntu does.
I've just created the Ubuntu 14.04 debian package, and I'm fighting with the Debian Wheezy package, but I think we should release the Ubuntu package already at least.

closed 2016-01-19 by fqez