JdeRobot/base #42: HowTo create a chroot environment on wiki

fqez, 2015-04-01:
Hi, 
I've just added a new page on the wiki describing the process to make a chroot environment in Debian/Ubuntu distributions. It's very simple, and you can find it [here](https://github.com/RoboticsURJC/JdeRobot/wiki/Create-a-chrooted-environment)
Any change is welcome!

Regards.

abhi-kumar, 2015-04-01:
@fqez could you please tell how a chroot environment will help testing the JdeRobot modules better than a normal sudo environment. I don't have much knowledge in the chroot domain, you explanation will be really helpful.

fqez, 2015-04-01:
Hi @abhi-kumar,

of course! Imagine you have your JdeRobot version well installed and working really fine in your system (i.e. Ubuntu 14.04 amd64), but you need to upgrade your ICE or OpenCV or PCL or whatever third-party library to a new version. You don't want to risk your well functional version of JdeRobot and it's dependencies removing the current version and installing the new one (this usually affects other packages and libraries), and you don't have time enough to make a whole fresh installation in other partition or virtual machine, so the fastest solution is to create a jail containing the same distribution of your main system (Ubuntu 14.04 amd64) with chroot and test JdeRobot with the new version of the library you need without touching your fine JdeRobot installation.
Realize that creating a chrooted environment in your machine makes your system believe that your root directory ("/") is in another place than the actual root of the system (like I explain on the wiki, the process in which you launch chroot believes that the root directory is in / while actually it is in /var/chroot/trusty_x64/, not letting you touch anything outside that directory and therefore not risking your current installation).
Another practical use for chroot is to test an especific program or library in a different distribution or architecture. For example, if you are working in Ubuntu 14.04 amd64 and you want to test if a library that you are using works fine in Debian Wheezy or Ubuntu 14.10 or Ubuntu 14.04 i386.

Hope it clarifies a bit your doubts.

Best!

abhi-kumar, 2015-04-01:
@fqez thanks for the explanation, it made everything clear now.

closed 2015-05-29 by fqez