JdeRobot/base #45: Add doc target and doxygen support

ojgarciab, 2015-04-08:
I'll implement document generation system.

There will a common target for every document system `doc` and a local dependency on every component/library.

lr-morales, 2015-04-08:
I have a [CmakeLists entry](https://svn.jderobot.org/users/lr.morales/pfc/trunk/src/components/viewer3D/build/CMakeLists.txt) and a [Doxyfile.in](https://svn.jderobot.org/users/lr.morales/pfc/trunk/src/components/viewer3D/build/Doxyfile.in) for a component which could be used as reference if you want.

ojgarciab, 2015-04-08:
I think that it is better to create a specific target that doesn't generate document in "all" target, only when "make doc" is executed.

Thanks for your Doxyfile.in, I'm working in a very generic one that is shared across all libraries and components. I'll do the commit as quick as I can.

ojgarciab, 2015-04-08:
I just implement an default doxyfile.in file that is used if there is not customized file in a library/component to avoid tenths of copies of the same generic Doxyfile.in file distributed across the project tree.

I have been defined some @...@ patterns to customize parameters on every library/component. They would be expanded to allow more customization.

@lr-morales the problem with your implementation is that whole project only has a unique "doc" target, so defining it several times on every component/library will throw a error like this:

```
CMake Error at actual_file (ADD_CUSTOM_TARGET):
  add_custom_target cannot create target "doc" because another target with
  the same name already exists.  The existing target is a custom target
  created in source directory
  "previous_dir/previous_file".
  See documentation for policy CMP0002 for more details.
```

With this implementation and CMakeLists.txt example every component/library will create its own target and later add a dependency on global "doc" target.

And as I said, "doc" target will generate documentation only when use "make doc" to avoid waste of time of regenerate documentation every time we compile the entire project or an library/component.

closed 2021-02-21 by jmplaza