JdeRobot/base #51: Add an IceDemo component for getting started with Ice and contributing to JdeRobot

hustcalm, 2015-04-11:
I'v been thinking about a good way to get started with JdeRobot components, after reading the source code, it comes to my mind that grasping the skeleton of JdeRobot is very essential. As JdeRobot is built upon `Ice`, so I think it is a good choice to add an `IceDemo` component to the source tree, this will need to deal with `Slice` and basic `Ice` server and client component.

The demo will follow the documentation [Ice Hello World Application](https://doc.zeroc.com/display/Ice/Hello+World+Application), the basic steps:

1. Write a Slice definition and compile it
2. Write a server and compile it
3. Write a client and compile it

Hopefully, by getting familiar with the demo component, new developers or contributors will soon know better of the source tree of JdeRobot and able to do some help. Also the demo will follow the programming style and coding conventions(see #22) once settled down, this will help developers to be better involved.

The task will be accomplished as:

- [ ] Add the demo component with Slice, server and client
- [ ] Compile and test the demo component
- [ ] Write good documentation for the demo

gsssrao, 2015-04-11:
@hustcalm 

Have you checked out this [link](https://zeroc.com/training/index.html). It consists of a full course and a student workbook along with slides explaining in detail ICE, Slice and even assignments to help new developers learn these concepts. If you are creating a wiki then it will help. It already has an assignment for creating Slice definition, writing server and other things.

hustcalm, 2015-04-12:
@gsssrao 

Hi, thanks for the hint. I will go through the training course and probably starting with the wiki pages is a good choice.

hustcalm, 2015-04-16:
After going through the training course, a [wiki page](https://github.com/RoboticsURJC/JdeRobot/wiki/Getting-Started-with-Ice) is just what we need. I will probably start and maintain the wiki page while developing the structure reconstruction component with Kinect 2. Thanks again for the awesome [link](https://zeroc.com/training/index.html). @gsssrao

closed 2015-04-16 by hustcalm

gsssrao, 2015-04-16:
You're welcome.