JdeRobot/base #52: camera_dump compilation fails.

fqez, 2015-04-13:
Hi!

I'm trying to compile the gazebo plugins in ~/jderobot/src/stable/components/gazeboserver/plugins/$plugin/
but when it comes to compile camera_dump.cc (both nao and pioneer plugins) the following error appears:

```
[ 68%] Building CXX object src/stable/components/gazeboserver/plugins/nao/CMakeFiles/camera_nao.dir/camera_dump.cc.o
/home/fran/JdeRobot/src/stable/components/gazeboserver/plugins/nao/camera_dump.cc: In function ‘void* myMain(void*)’:
/home/fran/JdeRobot/src/stable/components/gazeboserver/plugins/nao/camera_dump.cc:269:81: error: cannot allocate an object of abstract type ‘CameraI’
/home/fran/JdeRobot/src/stable/components/gazeboserver/plugins/nao/camera_dump.cc:87:7: note:   because the following virtual functions are pure within ‘CameraI’:
/home/fran/JdeRobot/src/stable/interfaces/cpp/jderobot/image.h:1364:37: note: 	virtual jderobot::ImageFormat jderobot::ImageProvider::getImageFormat(const Ice::Current&)
/home/fran/JdeRobot/src/stable/interfaces/cpp/jderobot/image.h:1367:18: note: 	virtual void jderobot::ImageProvider::getImageData_async(const AMD_ImageProvider_getImageDataPtr&, const string&, const Ice::Current&)
																	make[2]: *** [src/stable/components/gazeboserver/plugins/nao/CMakeFiles/camera_nao.dir/camera_dump.cc.o] Error 1
make[1]: *** [src/stable/components/gazeboserver/plugins/nao/CMakeFiles/camera_nao.dir/all] Error 2
make: *** [all] Error 2
```

I guess that the error is caused by the refactorization of the image.ice interface in ~/jderobot/src/stable/interfaces/jderobot. I've tried to fix the plugin with the current version of image.ice but I'm not able to make it compile. Any help?

Thanks in advance!

chanfr, 2015-04-14:
Hi,
you just need to implement a new method in your Camera "server" component and include a new argument to another method:

New method:

	virtual jderobot::ImageFormat getImageFormat(const Ice::Current& c)
	{
		return (mFormats);
	}
where mFormats is a 	
jderobot::ImageFormat mFormats; 

( you can define it in the private section of your class). 

And in the camera constructor you have to define the formats your components supports, if you are only using rgb8 without compression you can just include that format:
	mFormats.push_back(colorspaces::ImageRGB8::FORMAT_RGB8.get()->name);

The method you have to modify is:
getImageData_async.

	virtual void getImageData_async(const jderobot::AMD_ImageProvider_getImageDataPtr& cb, const std::string& format, const Ice::Current& c)

which now has a new argument (format). 

As you are only using a single format to your camera server you can just ignore this argument, or better checks if this fomat is the one that you supports, otherwise you will need to chance your replytask class to allow multiformat operations.

fqez, 2015-04-18:
Hi @chanfr 

I've tried your solution and it works fine for the quadrotor plugins. But camera_dump still fails...The error now is the following:

```
Linking CXX shared library libcamera_dump.so
/usr/bin/ld: ../../../../libs/visionlib/colorspaces/libcolorspacesmm.a(imagecv.cpp.o): relocation R_X86_64_32 against `pthread_cancel' can not be used when making a shared object; recompile with -fPIC
../../../../libs/visionlib/colorspaces/libcolorspacesmm.a: error adding symbols: Bad value
collect2: ld returned 1 exit status
make[2]: *** [src/stable/components/gazeboserver/plugins/pioneer/libcamera_dump.so] Error 1
make[1]: *** [src/stable/components/gazeboserver/plugins/pioneer/CMakeFiles/camera_dump.dir/all] Error 2
make: *** [all] Error 2
```

I've been testing and I've managed to make it compile in two different ways:

* Comment the line `` mFormats.push_back(colorspaces::ImageRGB8::FORMAT_RGB8.get()->name);`` : if I do this it compiles and binds the cameras, but it gives me a segmentation fault when I try to load the model in gazebo (pioneer model)
* Comment the colorspacesmm in the camera_dump target_link_libraries inside the CMakeLists.txt file of the pioneer plugin. : If I do this it compiles but it doesn't create the cameras so I can't use them when I load the model in gazebo (pioneer model).

Any suggestions?

Thank you so much :)

Although it compiles, when I try to load the model (pioneer) inside gazebo, it binds all other the plugins but camera_dump :S

rocapal, 2015-04-20:
@fqez I have just uploaded the patch to solve this issue. Just check it please, and tell me if you have some errors in your compilation and execution.

I also found the problem with libcolorspacesmm.a (just I removed the dependency in the CMakeList). I'm not sure about why it's trying to link against a non-static library.

fqez, 2015-04-20:
Hi @rocapal thanks for the update, but it doesn't work for me. I've found that solution this morning but despite of it compiles well and everithing is generated ok, when I load the pioneer model in gazebo, it binds every plugin perfectly but the cameras. It's like camera_dump doesn't exist anymore. I've checked the model.sdf and the model.config of pioneer and everything seems to be ok but there is not a single trace of camera_dump :(
The only way I found to make this work is doing the following:
* adding colorspacesmm to the camera_dump in the CMakeLists file
* commenting out the line ``mFormats.push_back(colorspaces::ImageRGB8::FORMAT_RGB8.get()->name);``
* the rest is excatly as your last update.

If I do so, it compiles everything ok aswell but in this case when I load the model in gazebo it DOES connect the cameras correctly and I am able to run introrob aswell (making a little change in the control.cpp class also).
Are you able to connect the cameras in gazebo?

jmplaza, 2015-04-21:
It seems a linking problem. Camera_dump needs libcolorspaces to get the name of the format of the images it serves: "colorspaces::ImageRGB8::FORMAT_RGB8.get()->name". In addition camera_dump generates a plugin code, that is, a dynamic library. Nevertheless libcolorspacesmm has been compiled as a static library (as indicated in its CMakeLists.txt, STATIC, so libcolorspacesmm.a is generated) and its code is not rellocatable. Maybe libcolorspacesmm should be created as dynamic library (libcolorspacesmm.so generated) and compiled using the -fPIC flag of the compiler --Position Independent Code-- as the error message suggests). This way its code will be rellocatable and will not overlap with other symbols that already use those addresses when 'introducing' it in the camera_dump.so. 

/usr/bin/ld: ../../../../libs/visionlib/colorspaces/libcolorspacesmm.a(imagecv.cpp.o): relocation R_X86_64_32 against `pthread_cancel' can not be used when making a shared object; recompile with -fPIC
../../../../libs/visionlib/colorspaces/libcolorspacesmm.a: error adding symbols: Bad value

JoseMaria

fqez, 2015-04-21:
@jmplaza hi, yep that's the main reason that makes this pluging doesn't compile. I've modified the colorspacesmm library to make it dynamic and it works like a charm. I've also refactored the camera_dump plugin and introrob component and now it works properly. I'm going to make a pull request with this changes in order to you to test it and tell me if it works for you too. The colorspacesmm thing... I don't know if we can make it dynamic so easily without breaking something else, but nowadays it works, as you said, JoseMaria.

cheers and thank you all!

fqez, 2015-05-19:
Ok this issue is solved adding colorspacesmmshare instead of colorspacesmm into the CMakeLists file in the camera_dump section. So I can close this!

closed 2015-05-19 by fqez