JdeRobot/base #54: opencvdemo component runtime error

abhi-kumar, 2015-04-17:
While running the opencvdemo component I get the following errors - 
"
Loading glade
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
"

Please help me cope with the issue.

Regards,
Abhishek

gsssrao, 2015-04-18:
Could you tell the contents of opencvdemo.cfg and cameraserver.cfg? In cameraserver.cfg are the endpoints this:
CameraSrv.Endpoints=default -h 0.0.0.0 -p 9999 
Also is the opencvdemo.cfg like this:
Opencvdemo.Camera.Proxy=cameraA:tcp -h 0.0.0.0 -p 9999

abhi-kumar, 2015-04-18:
cameraserver.cfg has "CameraSrv.Endpoints=default -h 0.0.0.0 -p 9999"

and opencvdemo.cfg has "Opencvdemo.Camera.Proxy=cameraA:tcp -h 127.0.0.1 -p 9998"

gsssrao, 2015-04-18:
Aah! That's the cause of the error. cameraserver is listening for client calls on port 9999 and opencvdemo is sending the call to port 9998. Change 9998 to 9999 and try.

abhi-kumar, 2015-04-18:
Sir, I changed the port setting as instructed and run the command " ./opencvdemo --Ice.Config=opencvdemo.cfg" but I still get the same error.

gsssrao, 2015-04-18:
Have you started the cameraserver. You need to first start the cameraserver and then when it is running, start opencvdemo on a different terminal.

PS: I am also a student. So you can drop the "Sir".

abhi-kumar, 2015-04-18:
Thanks a lot with that, I was assuming the fact that killing the terminal would still keep the server running as I recently studied about ICE and it has that option. We can implement that option I guess. Thanks for the valuable input from your side.

gsssrao, 2015-04-18:
You're welcome.

closed 2015-06-04 by chanfr