JdeRobot/base #76: JdeRobot packages for drone contest.

fqez, 2015-05-14:
Hi,

I'm creating this issue to let you give me feedback about the debian packages that I made for the next drone contest. Please leave here all the problems you face about the installation of JdeRobot from packages. 
You can see the instructions of the installation [here](http://jderobot.org/Downloads#Debian_packages_from_our_APT_Repository)

Regards!

andresjhe, 2015-05-18:
Hi!
I tried the packages, and as far as I know, no bugs with quadrotor plugin for Gazebo. Introrob_py seems to be working fine, although the change camera button doesn't work, but I think this is to be expected for now from Alberto.

I tried introrob, but it's not working properly. Gazebo keeps looping this warning/error:
>
[Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Err] [ODEJoint.cc:495] SetParam(vel) boost any_cast error:boost::bad_any_cast: failed conversion using boost::any_cast
[Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:427] The 'fmax' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Wrn] [ODEJoint.cc:443] The 'vel' parameter is deprecated to enable Coulomb joint friction with the 'friction' parameter
[Err] [ODEJoint.cc:495] SetParam(vel) boost any_cast error:boost::bad_any_cast: failed conversion using boost::any_cast

Anyway, the installation process is flawless and it installs everything but introrob_py, which is being deal with in #81

fqez, 2015-05-18:
Hi @andresjhe! thanks for testing it :D hmmm those warnings are so strange... the plugins are suposed to be already refactored. 
Older gazebo versions support instructions like  ```SetMaxForce(index, value)``` or ```SetVelocity(index, value)``` (the ones that our plugins had before the refactorization). But now. for gazebo 4+, those instructions are deprecated and have been replaced by ```SetParam(key, index, value)``` where, <i>key</i> can take values like "fmax" or "vel" correspondig with the deprecated SetMaxForce and SetVelocity respectively. [here](https://groups.google.com/a/osrfoundation.org/forum/#!msg/gazebo/5wxVm6ghSlE/QX4FKMzmyX8J) and [here](http://answers.gazebosim.org/question/8371/kobukiturtlebot-doesnt-move-when-given-velocity/) you got examples about it.

About introrob_py, yes, it will be solved soon!
The change camera button does not work because there is only one camera connected to introrob_py, but as I said it will be fixed in the next package version.

I will give it a shot. Thank you again!

cbyte18, 2015-05-19:
I follow the install instruction and I install the enviroment With apt -get. Gazebo runs OK but I have a question , I have to install the developers version of joderobot ( git clone of jderobot ) ? or if I install the enviroment from apt -get the jderobot developer 's version is already installed ( where? )

cbyte18, 2015-05-19:
I answer myself. Yes 
I have to install following these steps http://jderobot.org/Manual-5#Compile_and_Install_JdeRobot_5_with_CMake

fqez, 2015-05-19:
Hi @cbyte18 I'm glad it works for you too!
If you mean to download the source code in order to compie the project from there, you can download it with the following command ```apt-get source jderobot``` and then follow the instructions [here](http://jderobot.org/Manual-5#Setting_up) you don't have to install anything else! only download the source code and do cmake and make! Don't pay attention at the Download the source code section, the last version of the code is here (GitHub) so the command I gave you before is enough.

Regards!

cbyte18, 2015-05-19:
I have downloaded the code cloning the git repository I did cmake and I have this error.

 /usr/include/libxml2
CMake Error at Deps/ice/CMakeLists.txt:25 (MESSAGE):
  *** IceStorm not found
Call Stack (most recent call first):
  CMakeLists.txt:96 (include)
The apt-get installation don't have installed the package, 
I found an error on the following steps http://jderobot.org/Manual-5#Install_dependencies the steps are for install the 3.4 version of Ice not the 3.5

Thanks

andresjhe, 2015-05-19:
Are you using Ubuntu 14.04+ or are you using any Ubuntu below that one? In case of 14.04+ you should install all Ice 3.5 dependencies

cbyte18, 2015-05-19:
I use ubuntu 14.04.05 

José Antonio Fernández Casillas

andresjhe, 2015-05-19:
Then substitute the 3.4 with 3.5 and try it again

fqez, 2015-05-19:
Hi, the Ice dependencies should be installed with the package, did you installed Ice 3.4 after the installation of the package? As I said before, you don't need to install anything else, the package already resolves all the dependencies. I'm gonna take a look at that error as soon as possible. 
You only have to install the package (as you did), download the code from git or from apt and type ```cmake .``` and ```make``` to compile the whole project. 
Make sure that you are using gcc-4.6 when compiling.

cbyte18, 2015-05-19:
I installed the Ice 3.5 after the error when I doing make of jderobot.

fqez, 2015-05-19:
It works now?

cbyte18, 2015-05-19:
No It don't woks yet noy I have the following error:

-- checking for module 'gtkglextmm-1.2'
--   package 'gtkglextmm-1.2' not found
CMake Error at /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:283
(message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:337
(_pkg_check_modules_internal)
  Deps/gtk2/CMakeLists.txt:10 (PKG_CHECK_MODULES)
  CMakeLists.txt:108 (include)

Notice I have the gcc version 4.6

josete@ubuntu:~/JdeRobot/build$ g++ --version
g++ (Ubuntu/Linaro 4.6.4-6ubuntu2) 4.6.4
Copyright (C) 2011 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

thanks

fqez, 2015-05-19:
Ok I have to upload a new version of the package this week anyway, so I will test deeply this time.I have done some quick test and it seems like there is something wrong with the installation. I will keep you updated of the changes with this issue. 
Thanks and sorry!

fqez, 2015-05-20:
Ok I know what is the problem here. The current package does not install developer version of our third party packages, it only installs the libraries themselves without the headers (needed to compile our project). That's why you cannot compile the source code, because the headers are not installed. In order to be able to compile the source, you might follow the instructions [here](http://jderobot.org/Manual-5#JdeRobot_dependencies) if you are in a hurry, or wait until I can create a developer version of the package (jderobot-dev) but this part if not sure yet!
The current package only install dependencies needed to launch components, not to develope them.

Cheers!

fqez, 2015-05-29:
There is a package out already! Jderobot-5.3.0-rc1. You can install it following [these instructions](http://jderobot.org/Downloads#Debian_packages_from_our_APT_Repository). For now this package is only available for Ubuntu 14.04 64-bits but there will be more soon!

cbyte18, 2015-05-30:
Thanks so much I’ll try it now

fqez, 2015-05-30:
You are welcome :) Remember, this package is only valid to launch components, not to compile JdeRobot from source! If you want to compile JdeRobot, you still have to install the dependencies manually!
[here](http://jderobot-developer-list.2315034.n4.nabble.com/JdeRobot-Debian-packages-ready-td4642687.html) you have futher information!

Cheers!

reysam93, 2015-06-04:
Hello!

I have installed jderobot following this instructions http://jderobot.org/Downloads#Debian_packages_from_our_APT_Repository and it doesn't create the folder .gazebo

I have also looked for the folder "programaciondrones" and for MyAlgorithm with  du | grep  programaciondrones  and du | grep MyAlgorithm   and I haven't found anything.

fqez, 2015-06-04:
Hi, have you tried to run gazebo just after installing the package? It's missing in the manual, but the first time you launch gazebo it will download the models from its database and create the needed folders such as .gazebo and .gazebo/models.

There is something missing in the packages right now. The installation does not copy some needed libraries in their own directories, so if you try to load a world with a drone in gazebo it won't work. I am working at it right now, so as soon as I have a fix, I will upload a new package.

fqez, 2015-06-04:
Ok the problem with the packages is solved. You can now install JdeRobot from packages with introrob_py included. Also, I've updated the manual with the instructions to set up gazebo. Tell me if you face any problem.

Regards.

reysam93, 2015-06-04:
Ok, but looking in the page of the champion-ship ( http://jderobot.org/Campeonato-de-drones ) I have realised that I still don't have the folder cfg inside .gazebo.

Also when I try to run gazebo gato_raton_1.world  it doesn't work. It only open gazebo and immediately close it.

fqez, 2015-06-04:
the cfg files are in /usr/local/share/jderobot/conf. You can run the following command from that directory:

```
gazebo ../gazebo/worlds/gato_raton_1.world
```

and it should work. Don't worry if gazebo crashes at launch, it happens a lot. Just retry opening it over and over until it works. I don't know why is that instable, but there's nothig to worry about :D 
You can also see the log file of gazebo when you try to load a world if you want to be sure. The log file is in: .gazebo/server-XXXXX/default.log

Good luck!

reysam93, 2015-06-06:
Now I have gazebo and the world running, but there is a problem importing the module Ice in sensor.py. When I try to run the component introrob with:
`python main.py`
 I get the next error:
samuel@ubuntu:~/jderobot/python/introrob_py$ python main.py
Traceback (most recent call last):
  File "main.py", line 21, in <module>
    from MyAlgorithm import MyAlgorithm
  File "/home/samuel/jderobot/python/introrob_py/MyAlgorithm.py", line 1, in <module>
    from sensors import sensor
  File "/home/samuel/jderobot/python/introrob_py/sensors/sensor.py", line 19, in <module>
    import sys, traceback, Ice
ImportError: No module named Ice

fqez, 2015-06-06:
Yes, today it will be a new package including these dependencies. Sorry for all the inconveniences. And thank you for the feedback. As soon I upload the new package I will post a message here.

Cheers!

fqez, 2015-06-06:
You can install these dependencies manually while I upload the package, so you can work inmediately. Just type:

```
sudo apt-get install python-zeroc-ice, python-opencv, python-qt4, python-qwt5-qt4 
```

reysam93, 2015-06-06:
Thanks.
I don't know how to use the drones with the introrob.
With the old introrob component it would be something like: introrob  --Ice.Config=ardrone.cfg
If thats the case I don't find the .cfg file.

Thanks again.

fqez, 2015-06-06:
Are you trying to run python introrob or c++ introrob? If you are trying to run the first one (you have to download it first), there is a .cfg file in its folder (introrob_py.cfg), If you are trying to launch the second one, all the cfg files for the jderobot project are in ```/usr/local/share/jderobot/conf```

reysam93, 2015-06-06:
I'm referring to the python introrob of this package, introrob_py.
When I run gazebo with the world gato_raton_1.world, and then I start the introrob component with:
python main.py,  I only get this massage:
samuel@ubuntu:~/jderobot/python/introrob_py$ python main.py
Interface camera not connected
Interface cmdVel not connected
Interface navdata not connected
Interface ardroneExtra not connected
Interface pose3D not connected

fqez, 2015-06-06:
Hi, you should type:
```
python main.py --Ice.Config=/usr/local/share/jderobot/conf/conf/introrob_py.cfg
```
But I recommend you to copy the .cfg file into your working directory, is more comfortable.
The .cfg file is located where I said before:
```
/usr/local/share/jderobot/conf
```

Regards.

fqez, 2015-06-07:
New package with .cfg files included!

andresjhe, 2015-06-07:
When I'm triying to launch gato_raton_1.world Gazebo stops working. Anyone else has this error? The same happens when from gazebo I deploy at the same time ArDrone and RedArDrone models into a new clean world. Introrob_py works fine with each of the ArDrone models for me.

fqez, 2015-06-07:
I think is a problem related with the referee plugin. If you try to comment out the plugin line in the gato_raton_1.world it should be fine. At least it worked for me. Try and tell to make sure what is the main problem.

andresjhe, 2015-06-07:
I have tried it too. I don't know why is happening because gazebo just closes shut after being paralyzed for few seconds. I'm going to try it with a non fresh installation I have on another PC

fqez, 2015-06-07:
Hmm yes, I am having the same issues with that. In another machine it works fine, so I think this is a pacakge generation problem. But without any trace of an error I don't even know what could be wrong. I am going to wait for the referee refactorization and see if it fixes something. Meanwhile I will keep looking for the problem.

andresjhe, 2015-06-07:
I have confirmed that in the packages both drones don't work in the same world. They can be launched separately but when trying to launch the second one, Gazebo freezes and closes. I'm opening a separate issue

reysam93, 2015-06-09:
When I  type:
~/jderobot/python/introrob_py$ python main.py --Ice.Config=/usr/local/share/jderobot/conf/introrob_py.cfg
I get this error:
Traceback (most recent call last):
  File "/home/samuel/jderobot/python/introrob_py/sensors/sensor.py", line 47, in __init__
    self.cameraProxy = jderobot.CameraPrx.checkedCast(basecamera)
  File "/usr/lib/python2.7/jderobot/camera_ice.py", line 191, in checkedCast
    return _M_jderobot.CameraPrx.ice_checkedCast(proxy, '::jderobot::Camera', facetOrCtx, _ctx)
ConnectionRefusedException: Ice.ConnectionRefusedException:
Connection refused

fqez, 2015-06-09:
try with ```introrob_py_simulated.cfg```

reysam93, 2015-06-09:
Same problem

fqez, 2015-06-09:
Well, can you post here the content of both .cfg files? and the output of gazebo when you launch the world?

reysam93, 2015-06-09:
samuel@ubuntu:~/jderobot/python/introrob_py$ cat introrob_py.cfg
Introrob.Camera.Proxy  = ardrone_camera:default -h 0.0.0.0 -p 9999
Introrob.Pose3D.Proxy  = ardrone_pose3d:default -h 0.0.0.0 -p 9998
Introrob.CMDVel.Proxy  = ardrone_cmdvel:default -h 0.0.0.0 -p 9995
Introrob.Navdata.Proxy = ardrone_navdata:default -h 0.0.0.0 -p 9996
Introrob.Extra.Proxy   = ardrone_extra:default -h 0.0.0.0 -p 9994

samuel@ubuntu:~/jderobot/python/introrob_py$ cat introrob_py_simulated.cfg 
Introrob.Camera.Proxy  = Camera:default -h localhost -p 9994
Introrob.Pose3D.Proxy  = ImuPlugin:default -h localhost -p 9000
Introrob.CMDVel.Proxy  = CMDVel:default -h localhost -p 9850
Introrob.Navdata.Proxy = Navdata:default -h 0.0.0.0 -p 9700
Introrob.Extra.Proxy   = Extra:default -h localhost -p 9701

Gazebo does't have any output when I launch the world.

fqez, 2015-06-09:
Well that's the problem Gazebo didn't found the libraries of the quadrotor and it didn't connect them. You should have a trace like 
```
fran@ubuntu:/usr/local/share/jderobot/conf$ gazebo ~/testquad/JdeRobot/src/stable/components/gazeboserver/worlds/gato_raton_1.world 
Constructor ToggleCamPlugin
Load: -629553704 cam_sensor_ventral On: 1. Camera Id: 1
Constructor ToggleCamPlugin
Load: 1933399104 cam_sensor_frontal On: 0. Camera Id: 0
Constructor ImuPlugin
ImuPlugin Endpoints > default -h localhost -p 9000
Constructor ToggleCamPlugin
Load: 1987889608 red_cam_sensor_ventral On: 1. Camera Id: 1
Constructor ToggleCamPlugin
Load: -1927819984 red_cam_sensor_frontal On: 0. Camera Id: 0
Constructor ImuPlugin
ImuPlugin Endpoints > default -h localhost -p 9100
Contructor QuadrotorPlugin
State: Initialize
Constructor StateController
State: Landed
Constructor PosePlugin
Quadrotor CMDVel Endpoints > default -h localhost -p 9850
Contructor QuadrotorPlugin
State: Initialize
Quadrotor Configuration Endpoints > default -h localhost -p 9997
Constructor StateController
State: Landed
Constructor PosePlugin
Quadrotor Camera Endpoints > default -h localhost -p 9994
Constructor CameraI
safeThread
Quadrotor Navdata Endpoints > default -h localhost -p 9700
Quadrotor CMDVel Endpoints > default -h localhost -p 9950
navdata start
Quadrotor Extra Endpoints > default -h localhost -p 9701
Quadrotor Configuration Endpoints > default -h localhost -p 9998
Quadrotor Camera Endpoints > default -h localhost -p 9995
Constructor CameraI
safeThread
Quadrotor Navdata Endpoints > default -h localhost -p 9800
navdata start
Quadrotor Extra Endpoints > default -h localhost -p 9801
```
Make sure you have the following files in ```/usr/local/share/jderobot/gazebo/plugins/quadrotor/```
```
fran@ubuntu:/usr/local/share/jderobot/gazebo/plugins/quadrotor$ ls
ardroneplugin.cfg  libposeplugin.so       redardroneplugin.cfg
imuplugin.cfg      libquadrotorplugin.so  redimuplugin.cfg
libimuplugin.so    libtogglecamplugin.so
```
If so, then make sure that you have that directory in you $LD_LIBRARY_PATH environment variable:
```
fran@ubuntu:/usr/local/share/jderobot/gazebo/plugins/quadrotor$ echo $LD_LIBRARY_PATH 
/usr/local/lib/jderobot/:/usr/lib/i386-linux-gnu/gazebo-5.1/plugins/:/usr/lib/x86_64-linux-gnu/gazebo-5.1/plugins/:/usr/local/share/jderobot/gazebo/plugins/pioneer:/usr/local/share/jderobot/gazebo/plugins/nao:/usr/local/share/jderobot/gazebo/plugins/kinect:/usr/local/share/jderobot/gazebo/plugins/quadrotor:/usr/local/share/jderobot/gazebo/plugins/refereeplugin:/usr/local/share/jderobot/gazebo/plugins/turtlebot
```
If you can't see the ```/usr/local/share/jderobot/gazebo/plugins/quadrotor```  in your $LD_LIBRARY_PATH then type:
```cat /usr/local/share/jderobot/gazebo/gazebo-setup.sh >> ~/.bashrc```
and
```source ~/.bashrc```

fqez, 2015-06-09:
By the way, there have been important changes related with quadrotors, so I highly recommend you to clone the repository now and try to do what I just told in the previous message! It's most likely that gazebo will crash if you try to launch gato_raton_1.world with the previous version of the repository!

reysam93, 2015-06-09:
Ok, I will clone the repository beacuse gazebo has crashed.
By the way, are you gonna put a new package with this changes?

fqez, 2015-06-09:
Yes, I'am working on it right now. when I have it I will post here!

fqez, 2015-06-09:
Ok the packages are ready! you can test it now.

Cheers!

closed 2016-01-19 by fqez