JdeRobot/base #77: The installation process not copy the quadrotor and refereeplugin to the installation directory

almartinflorido, 2015-05-15:
Hi,

I compiled and installed JdeRobot from sources, the installation process not copy the quadrotor and refereeplugin plugins to the installation directory (/usr/local/share/jderobot/gazebo/plugins). To solve it I rewrote the double quotes in CMakeLists.txt of gazeboserver component for these plugins. The line to modify is the next:

SET(PLUGINS "kinect" "nao" "pioneer" "quadrotor" "refereeplugin")

In my next pull request I'll upload the changes.

regards

almartinflorido, 2015-05-16:
The change of the quotes is not the solution, I am testing and the components are installed when running a second time and there have been changes in the CMakeLists.txt of gazeboserver component.

chanfr, 2015-05-20:
The detection of the compilled plugins were statically set before the plugins compillation so it will only work when the plugins has been compilled. 

Now the intallatkon is based on plugin targets and using a relative path so we can now use CMAKE_INSTALL_PREFIX to install the plugins in a customize path.

closed 2015-05-21 by almartinflorido