JdeRobot/base #80: Failed to load plugin libpgsplugin.so in Gazebo with the quadrotor plugin

almartinflorido, 2015-05-15:
Hi,

I compiled and installed Jderobot in a clean Ubuntu 14.04 x86 virtual machine, when I execute gazebo with the gato_raton_1.world the gazebo server log show the next:

Failed to load plugin libpgsplugin.so: libpgsplugin.so: cannot open shared object file: no such file or directory

To fix it I comment the libgpsplugin.so declaration in the model.sdf file for ArDrone and RedArDrone models.

Is necessary this plugin? where is it?

regards

[image]

DanielYS, 2015-05-17:
Hello,

the plugin libgpsplugin.so isn't necessary. You can comment the lines in the model.sdf file as you say. I'm going to correct the model file so this error message doesn't appear.

Regards

almartinflorido, 2015-05-17:
Thanks @DanielYS

closed 2015-05-17 by almartinflorido