JdeRobot/base #81: About the installation of python components

almartinflorido, 2015-05-15:
Hi,

I have some question about the python components:
1/ I found a cmake function to find depedencies for python components ( [1] ), my idea is to add the function to JdeRobot repository, is correct create a CMakeLists.txt with the function code under Deps/findPythonModules directory for that another dependencies can use this function?

2/ Now the installation process copy de executables, libraries, configurations files, ... to different installation directories. In python we don't have executables only .py files and other resource how .png, .ui files,... I think that we can create the directory /usr/local/share/jderobot/python to copy the python components and after we can create symbolic links from the principal .py of the component to /usr/local/bin. What do you think?

regards

[1]: https://github.com/ivansafrin/Polycode/blob/master/CMake/FindPythonModule.cmake

andresjhe, 2015-05-17:
About 2/ you can make a .py file an executable by adding at the top #!/usr/bin/python, then, you can execute it from a terminal like any other program in Linux. We can take advantage of that when we create the symbolic link from the principal .py and just call to the link without the command "python program.py"

almartinflorido, 2015-05-17:
Thanks @andresjhe  but with executable I meant compiled code not scripts (.py files), the question is the procedure to install python components. The actual installation process no copy .py files. My python component it is formed by various .py files and other resources, the idea is to discuss a method for installing these components in the system.

closed 2015-05-29 by almartinflorido