JdeRobot/base #92: New release of introrob.py

jmplaza, 2015-05-21:
Old release of introrob.py doesn't work after changing the image ICE interface. 

In addition, new improvements should be added in order to use it in the new drone programming course, like to work even without connection to CMDvel ICE interface, or ardrone-extra ICE. This will allow to connect introrob.py with a cameraserver to test the image processing from video files, for instance.

jmplaza, 2015-05-21:
Its integration in the JdeRobot building chain is also tricky. We should take into account the questions you commented in the mailing list about it, Alberto, introducing in CMake the dependencies of debian packages which contain required python modules for introrob.py (better than using PIP).

almartinflorido, 2015-05-24:
Uploaded in the pull request #98 , for the CMake chain I will create a new issue

commit 0db4cd3c (2015-05-25): Fix #92 - Added the new version of introrob_py

- In this version introrob_py can works without interfaces connected or just some connected. Moreover it added a color filter for programming drones course.
- Updated the gazebo-setup script with the new routes of the gazebo 5.1.
commit 9673d02f (2015-05-25): Merge pull request #98 from almartinflorido/issue_92

Fix #92 - Added the new version of introrob_py
closed 2015-05-25 by rocapal