JdeRobot/base #100: kinectplugin and flyingkinect are broken

jmplaza, 2015-05-25:
With the jump to Gazebo-5 and the new camera interface these two plugins are broken. They don't work with the new release of the simulator.

chanfr, 2015-05-25:
I've updated the kinect plugin (issue #95). Where is the code of flyingkinect?

jnbosgos, 2015-05-25:
Hi, the code of flyingkinect is in http://svn.jderobot.org/users/jnbosgos/pfc-teleco/trunk/gazebo/plugins/flyingKinect/

jmplaza, 2015-05-25:
Maybe we should add (through a pull-request from you @jnbosgos) the flyingkinect to the official git repository, and then refine it once it has been integrated into the JdeRobot source tree. This plugin provides RGBD data and allow to move the 3D position of the simulated sensor in the Gazebo world.

fqez, 2015-05-25:
It gives me a warning when compiling the kinect plugin. Is related with gcc version. But it's just a warning though.

```
Scanning dependencies of target kinectPlugin
[ 74%] Building CXX object src/stable/components/gazeboserver/plugins/kinect/CMakeFiles/kinectPlugin.dir/kinectPlugin.cc.o
In file included from /usr/include/c++/4.6/backward/strstream:52:0,
                 from /usr/include/vtk-5.8/vtkIOStream.h:112,
                 from /usr/include/vtk-5.8/vtkSystemIncludes.h:40,
                 from /usr/include/vtk-5.8/vtkIndent.h:24,
                 from /usr/include/vtk-5.8/vtkObjectBase.h:43,
                 from /usr/include/vtk-5.8/vtkSmartPointerBase.h:26,
                 from /usr/include/vtk-5.8/vtkSmartPointer.h:23,
                 from /usr/include/pcl-1.7/pcl/visualization/point_cloud_geometry_handlers.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/point_cloud_handlers.h:41,
                 from /usr/include/pcl-1.7/pcl/visualization/common/actor_map.h:40,
                 from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/cloud_viewer.h:39,
                 from /home/2015-May/JdeRobot/src/stable/components/gazeboserver/plugins/kinect/kinectPlugin.h:29,
                 from /home/2015-May/JdeRobot/src/stable/components/gazeboserver/plugins/kinect/kinectPlugin.cc:18:
/usr/include/c++/4.6/backward/backward_warning.h:33:2: warning: #warning This file includes at least one deprecated or antiquated header which may be removed without further notice at a future date. Please use a non-deprecated interface with equivalent functionality instead. For a listing of replacement headers and interfaces, consult the file backward_warning.h. To disable this warning use -Wno-deprecated. [-Wcpp]
```

closed 2015-06-04 by chanfr