JdeRobot/base #103: JdeRobot compilation fails

fqez, 2015-05-25:
Hi, there is something wrong with the current version of the repository because when I try to do ```make``` it gives me an error like this:

```
Scanning dependencies of target jderobotViewer
[ 37%] Building CXX object src/stable/libs/jderobotViewer/CMakeFiles/jderobotViewer.dir/JderobotViewer.cpp.o
In file included from /usr/include/boost/signals2/slot_base.hpp:23:0,
                 from /usr/include/boost/signals2/detail/tracked_objects_visitor.hpp:18,
                 from /usr/include/boost/signals2/slot.hpp:22,
                 from /usr/include/boost/signals2/connection.hpp:23,
                 from /usr/include/boost/signals2/signal.hpp:21,
                 from /usr/include/boost/signals2.hpp:19,
                 from /usr/include/pcl-1.7/pcl/visualization/boost.h:52,
                 from /usr/include/pcl-1.7/pcl/visualization/common/actor_map.h:48,
                 from /usr/include/pcl-1.7/pcl/visualization/pcl_visualizer.h:48,
                 from /home/2015-May/JdeRobot/src/stable/libs/jderobotViewer/JderobotViewer.h:15,
                 from /home/2015-May/JdeRobot/src/stable/libs/jderobotViewer/JderobotViewer.cpp:8:
/usr/include/boost/variant/variant.hpp: In member function 'void boost::variant<T0, T1, T2, T3, T4, T5, T6, T7, T8, T9, T10, T11, T12, T13, T14, T15, T16, T17, T18, T19>::move_assigner::internal_visit(RhsT&, int) [with RhsT = boost::shared_ptr<void>, T0_ = boost::shared_ptr<void>, T1 = boost::signals2::detail::foreign_void_shared_ptr, T2 = boost::detail::variant::void_, T3 = boost::detail::variant::void_, T4 = boost::detail::variant::void_, T5 = boost::detail::variant::void_, T6 = boost::detail::variant::void_, T7 = boost::detail::variant::void_, T8 = boost::detail::variant::void_, T9 = boost::detail::variant::void_, T10 = boost::detail::variant::void_, T11 = boost::detail::variant::void_, T12 = boost::detail::variant::void_, T13 = boost::detail::variant::void_, T14 = boost::detail::variant::void_, T15 = boost::detail::variant::void_, T16 = boost::detail::variant::void_, T17 = boost::detail::variant::void_, T18 = boost::detail::variant::void_, T19 = boost::detail::variant::void_]':
/usr/include/boost/variant/detail/visitation_impl.hpp:130:9:   instantiated from 'typename Visitor::result_type boost::detail::variant::visitation_impl_invoke_impl(int, Visitor&, VoidPtrCV, T*, mpl_::true_) [with Visitor = boost::variant<boost::shared_ptr<void>, boost::signals2::detail::foreign_void_shared_ptr>::move_assigner, VoidPtrCV = void*, T = boost::shared_ptr<void>, typename Visitor::result_type = void, mpl_::true_ = mpl_::bool_<true>]'
```

and so on. I tried to compile an older repository version and it worked, so I think there is something broken in the latest updates.

rocapal, 2015-05-25:
I don't have this problem in Debian sid with gcc-4.9. Can you check it which version of gcc do you have?

But if the previous version compiles well, for sure we have an issue. Can you see the differences between the previous version and the last one?

fqez, 2015-05-25:
I am using gcc-4.6 in ubuntu 14.04. The only difference with the jderobotviewer is the last commit of its CMakeLists.txt issue #89 
The other libraries (parallelIIce, jderobotutil, depthlib, etc) seems to compile ok, but when it comes to jderobotviewer it crashes.
I am testing the Daniel Yague version of git (it has not the lastest commits) ant it wokrs fine.

rocapal, 2015-05-25:
May be it's other issue, because in our testing box (for continous integration) it's working well too.

fqez, 2015-05-25:
yes, probably, but the only change it has is the shown in that issue, Just for information :)

fqez, 2015-05-25:
It works with gcc-4.8 though! So I suppose that it is an issue with the gcc version.

chanfr, 2015-05-25:
Hi,

it seems that boost doesn't works when linking while using c++0x compatibility with a gcc version lower than 4.7. So I've include this check in the main cmakelists.

closed 2015-05-25 by chanfr

fqez, 2015-05-25:
Oh ok! thank you Fran!

fqez, 2015-05-25:
Yes it's working now. Thank you again!