JdeRobot/base #104: ArDrone and RedDrone possible Gazebo 5 plugin memory leak?

andresjhe, 2015-05-26:
I have been refactoring the referee for the drone contest and I noticed when leaving Gazebo open for testing purposes, that my ram kept getting own by the Gazebo process. This only happened when having the world with these two models open. When my computer started to use swap space Gazebo started to show this message everytime:
> [Wrn] [Publisher.cc:131] Queue limit reached for topic /gazebo/default/pose/local/info, deleting message. This warning is printed only once.

I don't know if this message is related or not, but is the only one that posted when launching gazebo --verbose.

DanielYS, 2015-06-06:
I get that message even if I delete the plugins from the ArDrone SDF file, but I see how memory use increase all the time with the plugins being executed.

I think this issue is related with memory allocation in the velocitycallback function on quadrotorplugin.cc. I've moved some lines to the load method and it seems that memory use doesn't increase anymore, so I've made a pull request.

andresjhe, 2015-06-09:
Memory leak is still present. When both drones are in the same world is even more noticeble.
When a single one is present, it raises slower, but still exists.
I couldn't trace it yet

varhub, 2015-12-19:
**bug was characterized: miss usage of cv::Mat**
cv::Mat is already a smart pointer. So you may never do a new operation.
(This is way cv::Mat has clone() operation, because copy is almost reference copy).

Move new to Load just reduces memory leak to 1 per execution

Related to #216
Bug not present at new component quadrotor2

jmplaza, 2016-03-09:
Solved in the solution for #229

closed 2016-03-09 by jmplaza