JdeRobot/base #110: CMake warning messages: imported target relating to VTK references to non-existed files

hustcalm, 2015-05-29:
The `cmake .` command gives many messages regarding non-existed files related to VTK imported target, the outputs are as following:

```bash
version:4.8.2
/home/lihang/gsoc2015/JdeRobot/src/stable/libs/colorspaces
/home/lihang/gsoc2015/JdeRobot/src/stable/libs/jderobotutil
/home/lihang/gsoc2015/JdeRobot/src/stable/libs/progeo
/home/lihang/gsoc2015/JdeRobot/src/stable/libs/pioneer
/home/lihang/gsoc2015/JdeRobot/src/stable/libs/fuzzylib
/home/lihang/gsoc2015/JdeRobot/src/stable/libs/visionlib
/home/lihang/gsoc2015/JdeRobot/src/stable/libs/parallelIce
*** GAZEBO LIBRARIES FOUND: /usr/include/gazebo-5.1;/usr/include/sdformat-2.3 - /usr/include/OGRE
*** Fireware ok
-- The imported target "vtkWrapTcl" references the file
   "/usr/bin/vtkWrapTcl"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/vtk-5.8/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkWrapTclInit" references the file
   "/usr/bin/vtkWrapTclInit"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/vtk-5.8/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkWrapPython" references the file
   "/usr/bin/vtkWrapPython"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/vtk-5.8/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkWrapPythonInit" references the file
   "/usr/bin/vtkWrapPythonInit"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/vtk-5.8/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkParseJava" references the file
   "/usr/bin/vtkParseJava"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/vtk-5.8/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkWrapJava" references the file
   "/usr/bin/vtkWrapJava"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/vtk-5.8/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkCommonTCL" references the file
   "/usr/lib/libvtkCommonTCL.so.5.8.0"
but this file does not exist.  Possible reasons include:
[... 606 lines cut ...]
-- Processing component rgbdViewer
-- Processing component giraffeClient
-- Processing component rgbdManualCalibrator
-- Processing component naoviewer
-- Processing component kinect2server
****LIBS: /usr/local/lib/libfreenect2.sopthread
****INC: /usr/local/include/usr/local/include/tinythread
-- Processing component replayer
-- Processing component gazeboserver
-- Processing plugin quadrotor
-- Processing plugin nao
-- Processing plugin refereeplugin
-- Processing plugin kinect
-- Processing plugin pioneer
-- Processing component basic_component
-- Processing component colorTuner
-- Processing component openniServer
-- Processing component basic_server
-- Processing component pclRGBDServer
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lihang/gsoc2015/JdeRobot
```

Clearly the warning messages are gone, however, I can't figure out what exactly is the root cause. Hope that someone can come up with a solution.

chanfr, 2015-06-04:
Hi,
if you remove  include(${DEPS_DIR}/pcl/CMakeLists.txt) from the main CMakeLists it means that you are not "loading" PCL so all the components that need PCL will be disabled. 

Those warning come from vtk, PCL only uses a set of vtk submodules but uses vtk cmake to resolve the dependencies. Vtk cmake tries to find all the modules so each warning references an unresolved module but we dont need them. 

I dont know how to avoid those warnings, only installing all the vtk modules....

regards,
Fran.

fqez, 2015-06-04:
Yes, it does not affect at our compilation at all. I've been doing some research time ago, but I didn't find anything to avoid these warnings. The only way I think we can get rid of them is, as Fran said, installing all of the modules. But it's not necessary I guess.

Regards.

hustcalm, 2015-06-04:
@chanfr @fqez 
OK, no big deal then, I will try to see if there are options which can pass to `cmake` to suppress these warning. Thanks for your answers. 

Let's keep this issue open for now and see if we can come with a good solution some time.

fqez, 2016-07-25:
There is no way to overcome this atm, so I close this. We are in ways to update some third party dependencies versions like Ice 3.6, gazebo 7, vtk 6, etc. So I close this untill that update. If that persist, we can reopen this issue.

closed 2016-07-25 by fqez