JdeRobot/base #123: basic_component won't exit when using GTK and throw invalid pointer when using Qt after closing the main window

hustcalm, 2015-06-04:
Hi all,

When trying basic_component(use replayer as the Ice server to serve images) to learn JdeRobot development, the following issues pop up:

[image]

As can be seen from above, the application won't exit after closing the main window when using GTK gui,    one has to press `Ctrl+C` to force quit. When using Qt gui, the application can exit, however it throws an invalid pointer error and then crashes.

When using `off` as the gui option, the application starts normally but prints nothing to the terminal. Though this is not a bug, however it's better that at least one warning message given to make the users not curious.

What is your idea?

Lihang

fqez, 2015-06-04:
Hi Lihang,

yes this is a known issue but I had no time for fix it, sorry. The GTK issue can be easily fixed, but the Qt one is more tricky. I will take a look of it when I have some free time, and I will tell you what's is wrong. 
You are right, that component should have a warning when you launch it without GUI, thanks for the advice and sorry for the inconvenience.

Fran.

hustcalm, 2015-06-04:
Hi Fran,
@fqez OK, I will also try to do some help if possible. Never mind, really a good component to get started with writing new components for JdeRobot.

Lihang

closed 2016-06-22 by fqez