JdeRobot/base #124: CMake error when processing introrob_py

hustcalm, 2015-06-04:
Hi all,

The new commit [Fix #106 - Python components installation](https://github.com/RoboticsURJC/JdeRobot/commit/37c574a1d0072c243ded9dd5ce6382ed8be3ef15) seems to bring troubles when running `cmake .`, the outputs on my machine(Ubuntu 14.04 LTS) are as follows:

[image]

What could be wrong?

Lihang

chanfr, 2015-06-04:
I can't reproduce the problem but can you try to remove the ${} on each variable?

I mean changing:
${ICE_PY_MODULES} AND ${NUMPY_MODULES} AND ${OpenCV_PY_MODULES} AND ${QT4_PY_MODULES} AND ${QWT5_PY_MODULES}

to

ICE_PY_MODULES AND NUMPY_MODULES AND OpenCV_PY_MODULES AND QT4_PY_MODULES AND QWT5_PY_MODULES

hustcalm, 2015-06-04:
Yeah it works!!! @chanfr  Could this issue be related with the cmake version? I'm using `2.8.12.2`.

[image]

[image]

chanfr, 2015-06-04:
Great! 
yes maybe the problem is the cmake version but anywayt it is better to check variables with dependencies without the ${} as we are checking the definition of the variable not its content. 

Can you send a pull request with the solution?

thanks!

hustcalm, 2015-06-04:
Sure, see #125 . @chanfr

commit 2220e105 (2015-06-04): Fix #124: CMake error when processing introrob_py
commit f068dadf (2015-06-04): Merge pull request #125 from hustcalm/fix_cmake_variable

Fix #124: CMake error when processing introrob_py
closed 2015-06-04 by rocapal