JdeRobot/base #128: Some quadrotor configuration files missing when installing gazebo from packages.

fqez, 2015-06-04:
Hi,

there is a problem with some .cfg files in the installation of jderobot. The following 3 files are missing:

* referee.cfg
* redimuplugin.cfg
* redardroneplugin.cfg

This causes problems when trying to launch gato_raton_1.world with gazebo:

```
fran@ubuntu:/usr/local/share/jderobot/conf$ gazebo ../gazebo/worlds/gato_raton_1.world 
Constructor ToggleCamPlugin
Load: 330940040 cam_sensor_ventral On: 1. Camera Id: 1
Constructor ToggleCamPlugin
Load: 827936016 cam_sensor_frontal On: 0. Camera Id: 0
Constructor ImuPlugin
Constructor ToggleCamPlugin
Load: 598990456 red_cam_sensor_ventral On: 1. Camera Id: 1
Constructor ToggleCamPlugin
Load: -1813277208 red_cam_sensor_frontal On: 0. Camera Id: 0
Constructor ImuPlugin
ImuPlugin Endpoints > default -h localhost -p 9000
     PropertiesI.cpp:407: Ice::FileException:
     file exception: No such file or directory
     path: redimuplugin.cfg
     PropertiesI.cpp:407: Ice::FileException:
     file exception: No such file or directory
     path: referee.cfg
Contructor QuadrotorPlugin
State: Initialize
Constructor StateController
State: Landed
Constructor PosePlugin
Quadrotor CMDVel Endpoints > default -h localhost -p 9850
Quadrotor Configuration Endpoints > default -h localhost -p 9997
Quadrotor Camera Endpoints > default -h localhost -p 9994
Contructor QuadrotorPlugin
State: Initialize
Constructor StateController
State: Landed
Constructor PosePlugin
Constructor CameraI
     PropertiesI.cpp:407: Ice::FileException:
     file exception: No such file or directory
     path: redardroneplugin.cfg
safeThread
Quadrotor Navdata Endpoints > default -h localhost -p 9700
navdata start
Quadrotor Extra Endpoints > default -h localhost -p 9701
```

Looking at the /usr/local/share/jderobot/conf directory where they are supposed to be, I can't see these files. And in the source code they are missing too. Someone has them?

Regards,
Fran.

almartinflorido, 2015-06-06:
Done!

andresjhe, 2015-06-06:
I though I did a pull request refactoring the referee, but it seems it's not in the git.
I will pull request again trying to fix again the referee. The cfg file is not longer needed. I will also modify the worlds so it will work with the refactored referee component

fqez, 2015-06-07:
Well, they are already included with the lastest pull request of Alberto. There is even a package which I've just uploaded. I won't close this issue until Andrés refactor this component.

Thanks!

fqez, 2015-06-09:
The referee is already refactored too, so I close this issue. 

Thanks!

closed 2015-06-09 by fqez