JdeRobot/base #146: CMake Error: source directory plugins/refereeplugin does not contain a CMakeLists.txt file

hustcalm, 2015-06-11:
[image]

jmplaza, 2015-06-11:
uhm... that plugin for the referee of the drone contest is not needed anymore, should be deleted from the source tree.

fqez, 2015-06-11:
That plugin is not in the source code anymore. Please download the lastest version.
You should not have the refereeplugin in gazeboserver/plugins/

Regards.

rocapal, 2015-06-11:
Right, it's a cmake cache problem. Just clean your workspace or download the lastest version in a clean workspace, as fqez suggested you.

hustcalm, 2015-06-11:
Thanks for all the answers, my bad.

A `rm -rf CMakeCache.txt` will do the trick, then `cmake .` will perform normally.

closed 2015-06-11 by hustcalm