JdeRobot/base #156: Allow the load of multiple Kinects in Gazebo

fqez, 2015-06-30:
There is the need to load multiple kinect models in Gazebo with their own endpoints. This is usefull if we want to track people in an environment due to we need more than one sensor to make the job. I almost got a solution about this issue, so the pull request is comming!

closed 2015-10-30 by fqez

varhub, 2015-11-10:
Hello @fqez,

Could you explain some day how you overcame this issue?
Also, any additional info about gazebo plugins would be much appreciated.

fqez, 2015-11-10:
Hi @varhub, of course!

The changes made here mainly focus on the ICE communication thread of each plugin. Gazebo will create this thread once from [here](https://github.com/fqez/JdeRobot/blob/f52090a299fce4b63711e9e3843c82b004dede09/src/stable/components/gazeboserver/plugins/flyingKinect/kinect.cc#L83)

```
[...]
void DepthCameraPlugin::OnNewDepthFrame(const float *_image,
		unsigned int _width, unsigned int _height,
		unsigned int _depth, const std::string &_format)
{
	if ( count == 0 )
	{
		count++;
		std::string name = this->parentSensor->GetName();
		nameKinect = std::string("--Ice.Config=" + name + ".cfg");
		pthread_t thr_kinect;
		pthread_create(&thr_kinect, NULL, &mainKinect, (void*)this);
[...]
```
And once the plugin is binded, it will send or receive information, depending on the client needs. In the meantime, the rest methods of gazebo will keep working by its own without caring of the communication thread (Gazebo main loop), creating images in this specific case. So Gazebo only exchange information through ICE communications with the nested classes (KinectI, CameraI and CameraII)

```
Ice::ObjectPtr object = new KinectI(std::string("pointcloud1"),  kinect);
Ice::ObjectPtr object2 = new CameraI(std::string("cameraRGB"),  kinect,colorspaces::ImageRGB8::FORMAT_RGB8.get()->name);
Ice::ObjectPtr object3 = new CameraII(std::string("cameraDepth"),  kinect, colorspaces::ImageRGB8::FORMAT_DEPTH8_16.get()->name);
```
**And the changes are**
Instead of create a single object adapter with an specific endpoint

```
Ice::ObjectAdapterPtr adapter = ic->createObjectAdapterWithEndpoints("Kinect", Endpoints);
```

I've created a list of possible endpoints took from the .cfg file

```
std::vector<std::string> endpoints;        
std::string property = "";
for (int i=0; i<6; i++) {
    property = "Kinect"+boost::to_string (i)+".Endpoints";
    std::string Endpoints = prop->getProperty(property);
    endpoints.push_back(Endpoints);
```

the .cfg file has this format:

```
Kinect0.Endpoints=default -h localhost -p 9997
Kinect1.Endpoints=default -h localhost -p 9998
Kinect2.Endpoints=default -h localhost -p 9999
Kinect3.Endpoints=default -h localhost -p 10000
Kinect4.Endpoints=default -h localhost -p 10001
Kinect5.Endpoints=default -h localhost -p 10002
Kinect6.Endpoints=default -h localhost -p 10003
```

So now, we can tell the plugin that it can look for an unbinded endpoint in to create an ICE connection.

```
while (!connected) {
    try {
        adapter = ic->createObjectAdapterWithEndpoints("Kinect", endpoints[cont]);
	connected = true;
	std::cout << "Connected to free endpoint: " << endpoints[cont] << std::endl;
    }catch (Ice::SocketException& ss) {
	cont++;  // to look for the next endpoint in the endpoints list.
    }
}
```

This block does the exact same thing that the prior version of the plugin, but I've surrounded the object adapter creation sentence with a try-catch block.
The try-catch block, tries to create and object adapter to bind with taking an specific endpoint from the endpoints list, so it will iterate the list trying to bind with an endpoint till eventually it reach one free. If there is no free endpoints left, it will throw an exception as usual in the prior version of the plugin (with only one connection allowed per plugin).

Of course this is not the best solution, but it is the simplest one I guess.

varhub, 2015-11-10:
Understood @fqez!
One thread for kinect and each one fight to obtain lower ID (=port).
*Ice related*, each kinect has it own context (Ice::Communicator) and adapter.

And at DepthCameraPlugin::OnNewDepthFrame()
`count==0` is for initialization purposes.

I think I need a *master class* about gazebo plugins, so if you want to...

PS: should  CV_32FC1 be CV_F32C4? (at [kinect.cc#L135](https://github.com/fqez/JdeRobot/blob/f52090a299fce4b63711e9e3843c82b004dede09/src/stable/components/gazeboserver/plugins/flyingKinect/kinect.cc#L135))

fqez, 2015-11-10:
Glad of it @varhub :)
Whenever you want I can explain the little knowledge that I have about gazebo plugins!

Why do you say it should be a 4-channel (CV_32FC4) image? This is a depth image (grayscale), it does not have the 3 RGB components, right? Or I'm missing something? :S 
See http://stackoverflow.com/a/26331888/5467655

varhub, 2015-11-10:
No, it is just about magic numbers.
See it:
```
image.create(_height, _width, CV_32FC1);
memcpy( (float *)image.data,(float *) _image, _width*_height*4 );
```
**_width * _height * 4**

Is **4** sizeof(float) or r,g,b,d channels?
sizeof(float) or course, but I got confused

For consistency, you should use same type that OpenCV, otherways moving from 32 to 64 bits or far away will be buggy. So *float* itself can not be used (float32_t?).
Conclusion: forget it at all