JdeRobot/base #172: Installation section of the manual is not updated and mixed with Downloads

jmplaza, 2015-07-24:
Installation instructions are bad located and mixed with Download info in the mediawiki. In the download section several releases are mixed.

fqez, 2015-07-24:
I can carry that issue. How do you exactly want the mediawiki to be? I think I messed up the Download section with the hurry of the drones course, but now we have time I can improve the contents of the whole mediawiki!

jmplaza, 2015-07-27:
No problem. I have organized the manual a little bit further. 
1.- The dependency tables are now in the proper section of the manual (http://jderobot.org/Manual-5#JdeRobot_dependencies, that already existed). 
2.- The installation directions from debian package are now in the Installation section of the manual ( http://jderobot.org/Manual-5#Installing_JdeRobot_5), not in the Download section.

Further reviewing it is needed before closing this issue. 

In addition, there are many related improvements pending in the manual:
a.- distinction between stable releases and development release (the github as is). With svn every time we created a stable release we copied the development source tree in a different directory and that directory was left for bug corrections, if any. Maybe we have to think about how to organize it now that we use GitHub.
b.- The debian package of JdeRobot for developers (source JdeRobot) is missing, and the update of the corresponding directions in the Installation section of the manual.

closed 2016-01-19 by fqez