JdeRobot/base #173: ardrone_server/ cameraserver error in MK802IV (Armv7)

andresjhe, 2015-07-27:
Hi!
I've got this error when trying to get images from cameraserver (running on a MK802IV) to a PC running on a x86/x64 machine. 

The error I get on the MK802IV is this one:
>-! 07/27/15 14:28:34.087 warning: dispatch exception: /usr/include/Ice/BasicStream.h:175: Ice::UnsupportedEncodingException:
  > protocol error: unsupported encoding version: 1.1
   (can only support encodings compatible with version 1.1)
   >identity: cameraA
  > facet: 
  > operation: ice_isA
  > remote host: 192.168.1.5 remote port: 49810

The error I get on the PC is this one:

>/usr/local/share/jderobot/glade$ cameraview --Ice.Config=../conf/cameraviewmk.cfg 
>Loading glade
>Outgoing.cpp:499: Ice::UnknownLocalException:
unknown local exception:
>/usr/include/Ice/BasicStream.h:175: Ice::UnsupportedEncodingException:
protocol error: unsupported encoding version: 1.1
(can only support encodings compatible with version 1.1)
 > /usr/lib/libIce.so.34: IceInternal::BasicStream::throwUnsupportedEncodingException(char const*, int, unsigned char, unsigned char)
>  /home/linuxium/git/JdeRobot/src/stable/interfaces/cpp/jderobot/libJderobotInterfaces.so: IceInternal::BasicStream::startReadEncaps()
 > /usr/lib/libIce.so.34: Ice::Object::___ice_isA(IceInternal::Incoming&, Ice::Current const&)
  /home/linuxium/git/JdeRobot/src/stable/interfaces/cpp/jderobot/libJderobotInterfaces.so: jderobot::Camera::__dispatch(IceInternal::Incoming&, Ice::Current const&)
>  /usr/lib/libIce.so.34: IceInternal::Incoming::invoke(IceInternal::Handle<IceInternal::ServantManager> const&)

However, if I try to get images with a cameraserver running on a PC to the MK802IV I don't get this error. Both are running the same Jderobot version and if I try to run a cameraserver on the MK802IV and try to run also a cameraviewer it works.

Any suggestions? 
Thank you!

fqez, 2016-06-22:
Hi @andresjhe did you manage to make this work?

andresjhe, 2016-06-23:
I need to revisit it, this weekend I'll be able to.

Let me check it and I'll write you back.

closed 2021-02-21 by jmplaza