JdeRobot/base #174: new JdeRobot package (5.3) for developers

jmplaza, 2015-07-30:
A binary JdeRobot debian package was successfully created and uploaded to our repository. It allows the user to quickly install JdeRobot, Gazebo and run its components. Now a new package for developers should be prepared. It should 'install' the source code of JdeRobot and allow its modification, compilation, etc.. It will be very useful for people that want to develop new applications using our platform or people that want to modify an existing component.

fqez, 2015-10-30:
Being tested!

fqez, 2015-11-02:
Package out! You can see the installation instructions [here](http://jderobot.org/Manual-5#From_debian_packages_in_our_APT_Repository).

closed 2015-11-02 by fqez