JdeRobot/base #192: ArDrone's **roll** is inverted

varhub, 2015-10-09:
ARDrone's **roll** seems to be inverted.
Issue affects to real and simulated robot.

* Simulated robot test:
[image]

* Real robot:
[video](http://jderobot.org/store/varribas/uploads/videos/tfm-ardrone_server.webm) from min 1:20 to 1:40

**JdeRobot version**: github 834a1168895fd1f760eb604449c2453f7b0ee6a3 "#168 Included functionality to get a single image from jderobotViewer render"

PS: [unfetch changes from upstream](https://github.com/RoboticsURJC/JdeRobot/compare/834a1168895fd1f760eb604449c2453f7b0ee6a3...master) and [my changes](https://github.com/RoboticsURJC/JdeRobot/compare/834a1168895fd1f760eb604449c2453f7b0ee6a3...varhub:introrob_py-sandbox)

varhub, 2015-10-09:
Also roll control in inverted, so sign seems consistent.

Perhaps solution is switch sign in single line of code, but it can also be exposing a bug, like a different unpack or operation with an array due a numpy upgrade (for example).

Waiting for any kind of confirmation.

almartinflorido, 2015-10-11:
Hi varhub,

the pull request #197 modified the sign of roll.

varhub, 2015-10-22:
Hi @almartinflorido, just a question.
I did a minor refactor of Pose3D interface and notice that only roll is positive.
It is due a rotation of reference system or for quadrotor plugin compatibility?
https://github.com/varhub/JdeRobot/blob/d8c9c2b779ed8d63e99225052f8f109540bc3df1/src/stable/components/ardrone_server/src/components/pose3di.cpp#L69
Thanks in advance.

fqez, 2015-12-02:
Hi,
I've tested this same issue in a real ArDrone with introrob_py and both roll and yaw are inverted as well.

varhub, 2015-12-12:
added to #229 changeset

closed 2016-01-19 by fqez