JdeRobot/base #198: Inclusion in the third party repository (and packages) of Parrot SDK for their drones

jmplaza, 2015-10-12:
Our driver for ArDrones from Parrot (ArDrone1 and ArDrone2) wraps the Parrot SDK. We have done some modifications on the original SDK to work for us. This modified Parrot SDK should be included in our "Third party software", and a debian package should be generated for it. Our JdeRobot package should depend on this package.

fqez, 2015-10-12:
Yes, everything is ready for testing the SDK package now. This week the debian package will be uploaded to an apt repository to make it installable from command line.There is some information [here](http://jderobot.org/Varribas-tfm/ARDrone:cmake_advanced) about the manual installation of the SDK. The installation from packages will be posted there when the package has been tested.

varhub, 2015-10-23:
Task moved to RoboticsURJC/JdeRobot-ThirdParty/pull/1

varhub, 2015-10-27:
To finalize this issue, I managed ardronelib dependency through pkg-config.
Pull request at RoboticsURJC/JdeRobot-ThirdParty#3

It also includes a ffmpeg-0.8.pc to allow linking to this library (even from plain g++).

Adapted JdeRobot version at #203

I tested both and works like a charm. @fpez, could you test it?

---

PS: also a minor improvement at RoboticsURJC/JdeRobot-ThirdParty#2
For me, merge these three issues will conclude this one.

fqez, 2015-10-30:
Merged, thanks!

PR #203 (2015-10-30): Move to pkgconfig version of ardronelib
*Will close #198*

**Requires RoboticsURJC/JdeRobot-ThirdParty/pull/3**

Replace old style search, by pkg-config based.
Implied changes:
`PKG_SEARCH_MODULE(ardronelib OPTIONAL ardronelib)`

```
include_directories(${ardronelib_INCLUDE_DIRS})
link_directories(${ardronelib_LIBRARY_DIRS})
target_link_libraries(ardrone_server ${ardronelib_LIBRARIES})
```

*Tested.*
closed 2015-10-30 by fqez