JdeRobot/base #199: Discussion: Use single Ice Adapter

varhub, 2015-10-23:
Following recommendation from:
https://doc.zeroc.com/display/Ice35/Using+Multiple+Object+Adapters

*I suppose that share Ice adapter has been and introduced feature, but it seems to be available at least since ice3.4.*

Advantages:
* Less code:
  * Single createObjectAdapterWithEndpoints at code
  * Single xxx.Endpoints entry at Ice.Config
* Single used port

This means that connect to several robots will be easy

I let here an example with two adapters:
https://github.com/varhub/JdeRobot/commit/b3f430d21048b04c208038269111bf241456690f

fqez, 2016-06-22:
This would be interesting to talk about already, as well as #224 and #225. Nowadays is a pain to configure each execution of every example...

fqez, 2016-07-25:
I refactored the turtlebot plugin to allow this feature, now:

* quadrotor2
* flyingkinect2 
* turtlebot

 have only one Ice adapter.

closed 2021-02-21 by jmplaza