JdeRobot/base #202: Recorder and replayer support for pose3D interfaces

jmplaza, 2015-10-27:
Recorder should be extended to record data from pose3D interfaces too. And replayer should understand log files with such type of data too and provide them through a new pose3D interface. Yazmin and Daniel Azuara worked on this and built a working solution (http://svn.jderobot.org/users/ycumberbirch/pfc/trunk/). It should be integrated in the official source code of JdeRobot.

fqez, 2015-10-30:
I take care of this!

samartin, 2016-03-08:
@mazafrav is already working on this. He made a pull request (#341) but there was a problem compiling the replayer component. I think that in a few days this will be available.

fqez, 2016-03-08:
naming service running needed in order to make this works.

samartin, 2016-03-09:
In the new version that @mazafrav uploaded, the compilation error in the last message is gone. Now, it works running the namingService component first.

fqez, 2016-04-29:
Try my last PR. And launch replayer disabling namingService from configuration file. It should work!

samartin, 2016-05-02:
I tried recorder2 and replayer2 with the modifications of your last PR so namingService is no longer needed to successfully run replayer and it worked. I test it out with two cameras and a pose3D from flyingKinect and running the mapper3Drgbd and it did not work really well though. The position and the camera were even more out of sync than usual. I will investigate more about this tonight or tomorrow.

samartin, 2016-05-04:
I did a simple test to show that the images are displayed way faster than the position. In this simple test, I move the sensor a single position and then wait 2 or 3 seconds. It is [here](http://jderobot.org/store/samartin/uploads/videos/RecorderReplayerSyncProblem.ogv).

fqez, 2016-06-13:
I've revisiting this issue to try to find why there is so much delay between the position and the images, and I have no clue. The code seems to be ok, and if you check the timestamps for both pose3d info and image info, it has almost the same, for example (time in ms):

| instant | 0  | 1  | 2   | 3   | 4   | 5   | 6    |
|---------|----|----|-----|-----|-----|-----|------|
| image   | 24 | 72 | 266 | 460 | 656 | 851 | 1047 |
| pose3d  | 0  | 1  | 201 | 401 | 601 | 801 | 1002 |

So I have no idea why is this happening. Have you done several test or is this just an isolated case?

samartin, 2016-06-16:
That is weird. I think I tested it twice with the same result in both. I can try to do it again tomorrow to make sure I was doing it right. It would be great if it works because I would like to include some recorder and replayer test in my final year project thesis, which is almost finished.

samartin, 2016-06-17:
I tested it again using the last updates of the repository and it happens the same thing, the image always arrives first, like in the video. Sometimes it arrives with more delay and sometimes with less.

samartin, 2016-06-18:
I just found out what the problem is! It is this [line](https://github.com/RoboticsURJC/JdeRobot/blob/master/src/stable/tools/replayer2/replayer.cpp#L862). What I did is to remove the 1 extra second added to the ckeckState so it would end up like this: 
`long long int checkState=(pretime.toMicroSeconds())/1000;`
Because of it, the pose3D thread in replayer was being executed at 1 Hz and updating the pose3d info only each second.

In [this video](http://jderobot.org/store/samartin/uploads/videos/ReplayerFixed.ogv) a test can be seen. When I move the sensor with navigatorCamera, while I record, the position sometimes arrives sooner than the image and the pointCloud is calculated in a wrong way (this is because a gazebo issue). When the mapper3Drgbd is using the pose3D and image coming from replayer though, more images and positions out of sync can be seen than before, as the video shows. So it is not yet perfectly in sync, but almost.

fqez, 2016-06-21:
Merged the solution! Thank you very much!

closed 2016-06-21 by fqez