JdeRobot/base #207: CMake install not possible without root permissions

varhub, 2015-10-30:
#  CMake install not possible without root permissions
**Related to:** http://jderobot.org/Varribas-tfm/ARDrone:cmake_advanced#Bad_permissions
## Error
```
  file INSTALL cannot set permissions on
  "/tmp/Jderobot/build/_CPack_Packages/Linux/DEB/jderobot_5.3.0-rc1_amd64/core/usr/local/bin/introrob_py"

  file INSTALL cannot set permissions on
  "/tmp/Jderobot/build/_CPack_Packages/Linux/DEB/jderobot_5.3.0-rc1_amd64/core/usr/local/bin/refereeViewer"
```
## Hint
```
ls -l /tmp/Jderobot/build/_CPack_Packages/Linux/DEB/jderobot_5.3.0-rc1_amd64/core/usr/local/bin/introrob_py
---x--x--x 1 varribas varribas 76 Oct 30 11:51 /tmp/Jderobot/build/_CPack_Packages/Linux/DEB/jderobot_5.3.0-rc1_amd64/core/usr/local/bin/introrob_py
```

## Solution
1. Let default permissions
2. Add missed permissions

varhub, 2015-10-30:
**Let default permissions do not work. Same issue.**
Is a CMake bug?
```
INSTALL (FILES ${CMAKE_CURRENT_SOURCE_DIR}/introrob_py DESTINATION bin OPTIONAL)
```

Speed it up with per-component build
`cmake .. -Dbuild-default=OFF -Dbuild_introrob_py=ON -Dbuild_refereeViewer=ON`

varhub, 2015-10-30:
Problem also affects to **icestorm**

```
$ make package CMAKE_INSTALL_ALWAYS=1
CPack Error: Problem running tar command: /usr/bin/fakeroot "/usr/bin/cmake" -E tar cfz data.tar.gz  ./usr
CMake Error: archive_read_disk_entry_from_file: Can't open `./usr/local/bin/icestorm'
---x--x--x 1 varribas varribas 108 Oct 30 16:24 ./usr/local/bin/icestorm
```

varhub, 2015-10-30:
# Issue is harder than expected

Problem related to
* DESTINATION bin
* build into /tmp
* CMake bug

## Hint
Replace `DESTINATION bin` by `DESTINATION /tmp` will work:
```-rw-r--r-- 1 varribas varribas 76 Oct 30 14:33 _CPack_Packages/Linux/DEB/jderobot_5.3.0-rc1_amd64/tmp/introrob_py```

## Tried
1. Inspired in http://stackoverflow.com/questions/17381910/cmake-install-installing-config-files
Must binaries be marked as executable files with `COMPONENT binaries`?
*FAILED*

2. Use file(PROGRAMS instead FILES
> The PROGRAMS form is identical to the FILES form except that the default permissions for the installed file also include OWNER_EXECUTE, GROUP_EXECUTE, and WORLD_EXECUTE. This form is intended to install programs that are not targets, such as shell scripts. Use the TARGETS form to install targets built within the project.
> - https://cmake.org/cmake/help/v3.0/command/install.html
*FAILED*

3. Set all permissions
*FAILED*

4. Do no set any permissions
*FAILED*

fqez, 2015-10-30:
See this, it might help: http://cmake.3232098.n2.nabble.com/CPack-packages-are-missing-files-td4769882.html

varhub, 2015-10-30:
Tiro la toalla.
`cmake .. -Dbuild-default=OFF -Dbuild_introrob_py=ON && make package -j4 2>&1 | tee cmake.log`
```
Scanning dependencies of target logger
Scanning dependencies of target colorspacesmm
[  1%] [  2%] Building CXX object src/stable/libs/log/CMakeFiles/logger.dir/Logger.cpp.o
Building CXX object src/stable/libs/visionlib/colorspaces/CMakeFiles/colorspacesmm.dir/imagecv.cpp.o
Scanning dependencies of target geometry
[  3%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/matriz3x3.cpp.o
Scanning dependencies of target JderobotInterfaces
[  3%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/exceptions.cpp.o
[  4%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/cmdvel.cpp.o
[  5%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/matriz4x4.cpp.o
Linking CXX shared library libcolorspacesmm.so
[  5%] Built target colorspacesmm
Scanning dependencies of target xmlParser
[  6%] Building CXX object src/stable/libs/xmlParser/CMakeFiles/xmlParser.dir/xmlReaderParser.cpp.o
[  7%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/kinectleds.cpp.o
[  8%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/plano.cpp.o
Linking CXX shared library liblogger.so
[  8%] Built target logger
[  9%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/utils.cpp.o
[ 10%] Building CXX object src/stable/libs/xmlParser/CMakeFiles/xmlParser.dir/xmlWriterParser.cpp.o
[ 11%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/bodymovements.cpp.o
[ 12%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/Line2D.cpp.o
[ 13%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/Segment2D.cpp.o
[ 14%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/ptencoders.cpp.o
Linking CXX static library libxmlParser.a
[ 14%] Built target xmlParser
Scanning dependencies of target xmlParsershare
[ 15%] Building CXX object src/stable/libs/xmlParser/CMakeFiles/xmlParsershare.dir/xmlReaderParser.cpp.o
[ 16%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/Point2D.cpp.o
[ 17%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/Point3D.cpp.o
[ 18%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/genericData.cpp.o
[ 19%] [ 20%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/Line3D.cpp.o
Building CXX object src/stable/libs/xmlParser/CMakeFiles/xmlParsershare.dir/xmlWriterParser.cpp.o
[ 21%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/Plane3D.cpp.o
[ 22%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/datetime.cpp.o
Linking CXX shared library libxmlParser.so
[ 23%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/Segment3D.cpp.o
[ 23%] Built target xmlParsershare
[ 24%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/collada/colladaparser.cpp.o
[ 25%] [ 26%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/camera.cpp.o
Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/collada/color.cpp.o
[ 27%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/body.cpp.o
[ 28%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/collada/malla.cpp.o
[ 29%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/collada/material.cpp.o
[ 30%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/common.cpp.o
[ 31%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/containers.cpp.o
[ 32%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/image.cpp.o
[ 33%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/collada/submalla.cpp.o
[ 34%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/progeo/Progeo.cpp.o
[ 35%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/bodyencoders.cpp.o
[ 36%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/segmento.cpp.o
[ 37%] Building CXX object src/stable/libs/geometry/CMakeFiles/geometry.dir/math/recta.cpp.o
[ 38%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/jointmotor.cpp.o
[ 39%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/encoders.cpp.o
Scanning dependencies of target fuzzylib
[ 39%] Building C object src/stable/libs/fuzzylib/CMakeFiles/fuzzylib.dir/fuzzylib.c.o
Linking C static library libfuzzylib.a
[ 39%] Built target fuzzylib
[ 40%] Building CXX object src/stable/interfaces/CMakeFiles/JderobotInterfaces.dir/cpp/jderobot/ardroneextra.cpp.o
Linking CXX shared library libgeometry.so
[ 40%] Built target geometry
Scanning dependencies of target fuzzyshare
[ 41%] Building C object src/stable/libs/fuzzylib/CMakeFiles/fuzzyshare.dir/fuzzylib.c.o
Linking C shared library libfuzzylib.so
[ 41%] Built target fuzzyshare
Scanning dependencies of target jderobotutil
[... cut ...]
Please check /tmp/Jderobot/build/_CPack_Packages/Linux/DEB/Deb.log for errors
CPack Error: Problem compressing the directory
CPack Error: Error when generating package: jderobot
make: *** [package] Error 1

# Run command: /usr/bin/fakeroot "/usr/bin/cmake" -E tar cfz data.tar.gz  ./usr
# Working directory: /tmp/Jderobot/build/_CPack_Packages/Linux/DEB/jderobot_5.3.0-rc1_amd64
# Output:
CMake Error: archive_read_disk_entry_from_file: Can't open `./usr/local/bin/icestorm'
CMake Error: Problem creating tar: data.tar.gz
```

fqez, 2015-10-30:
if(NOT CMAKE_INSTALL_COMPONENT OR "${CMAKE_INSTALL_COMPONENT}" STREQUAL "Unspecified")
  list(APPEND CMAKE_ABSOLUTE_DESTINATION_FILES
   "/usr/local/bin/icestorm")
  if(CMAKE_WARN_ON_ABSOLUTE_INSTALL_DESTINATION)
    message(WARNING "ABSOLUTE path INSTALL DESTINATION : ${CMAKE_ABSOLUTE_DESTINATION_FILES}")
  endif()
  if(CMAKE_ERROR_ON_ABSOLUTE_INSTALL_DESTINATION)
    message(FATAL_ERROR "ABSOLUTE path INSTALL DESTINATION forbidden (by caller): ${CMAKE_ABSOLUTE_DESTINATION_FILES}")
  endif()
file(INSTALL DESTINATION "/usr/local/bin" TYPE FILE OPTIONAL PERMISSIONS OWNER_EXECUTE GROUP_EXECUTE WORLD_EXECUTE FILES "/home/fran/test2/JdeRobot/src/stable/components/icestorm/icestorm")
endif()

fqez, 2015-10-30:
I found a little bug in the root CMakeLists.txt. It seems that cmake is installing the same binaries twice from two different CMaksLists.txt.
The fragment I am talking about is the following (root CMakeLists):
```
FOREACH (currentBin ${LIST_COMPONENTS})
    #SET(EXIST EXIST-NOTFOUND)
    find_file(EXIST NAMES ${currentBin} PATHS ${CMAKE_CURRENT_SOURCE_DIR}/src/components/${currentBin})
    #IF (EXIST)
    INSTALL (FILES ${CMAKE_CURRENT_SOURCE_DIR}/src/stable/components/${currentBin}/${currentBin} DESTINATION /usr/local/bin OPTIONAL PERMISSIONS OWNER_EXECUTE GROUP_EXECUTE WORLD_EXECUTE COMPONENT core)
    #ENDIF(EXIST)
    #MESSAGE("${EXIST}")
ENDFOREACH(currentBin)
```

which collides with this one (~/Jderobot/src/stable/components/introrob_py/CMakeLists.txt):

```
	INSTALL (FILES ${CMAKE_CURRENT_SOURCE_DIR}/introrob_py DESTINATION bin OPTIONAL PERMISSIONS OWNER_EXECUTE GROUP_EXECUTE WORLD_EXECUTE WORLD_READ)
```
May it be the problem?

varhub, 2015-10-30:
One of these two lines must be deleted.
I presume that it must be from root CMakeLists.txt
Why?
1. `install (files ${CMAKE_CURRENT_SOURCE_DIR}/src/stable/components/${currentBin}/${currentBin}` is no longer valid. It also needs take into count CMAKE_BINARY_DIR
So we have 2 types of install:
    -  Precreated scripts - from source dir
    -  generated binaries - from binary dir
2. This permission specification is deprecated. To accept it, it must be labeled as `PROGRAMS` instead `FILES`
3. It is installing every component, even not marked to compiled. So build by components is broken by this sentence.
4. Hidden feature. It should be documented to avoid any future headaches.
5. Previously, a component was partially removed due this problem (icestorm), so this was happened at least twice (b2a6f6a5c2eb7393838a26658af6d0a8e3271219)

varhub, 2015-10-30:
After testing it with @fqez (I'm just a viewer through hangouts), we can clever explain it:
* Multiple install sentences over same file (or target) is allowed, but it will only take first on collision
* `PERMISSIONS OWNER_EXECUTE GROUP_EXECUTE WORLD_EXECUTE` is deprecated and break it all, please do not use it.
* Use `PROGRAMS` to avoid set explicitly permissions. At now, it set to 755.

These tips will fix this issue.

commit acfd2278 (2015-10-30): Fix #207: Using PROGRAMS and removed PERMISSIONS
closed 2015-10-30 by fqez