JdeRobot/base #211: VisualHFSM generates python code

reysam93, 2015-10-31:
Hello, I continue working in visualHFSM, and now I am adding the possibility of generate python code, but it still doesn't has a runtime GUI, but I'm working on it.

varhub, 2015-11-02:
Hello @reysam93 
I suppose that is too late, but I tell you anyway.

Instead of use std::cout and mix *plain* code with *variables*, you can use **printf** instead.

Example I:
```
void Generate::generateMain_py (){
	printf(
"if __name__ == '__main__':\n"+
"\n"+
"	signal.signal(signal.SIGINT, signal.SIG_DFL)\n"+
"	automata = Automata()\n"+
"	try:\n"+
"	automata.connectToProxys()\n"+
	/*TODO more automatagui*/
"		automata.start()\n"+
"       automata.join()\n"+
"\n"+
"       sys.exit(0)\n"+

"   except:\n"+
"       traceback.print_exc()\n"+
"       automata.destroyIc()\n"+
"       sys.exit(-1)\n"
    );
}
```

Example II:
```
void Generate::generateConnectToProxys_py(){
	printf(
"	def connectToProxys(self):\n"+
"		self.ic = Ice.initialize(sys.argv)\n"+
"\n"
	);
	for ( std::list<IceInterface>::iterator listInterfacesIterator = this->listInterfaces->begin();
			listInterfacesIterator != this->listInterfaces->end(); listInterfacesIterator++ ) {
		const char* iceName = listInterfacesIterator->getName().c_str();
		const char* iceInterface = listInterfacesIterator->getInterface().c_str()
		printf(
"		# Contact to %s\n"+ /*iceName*/
"		%s = self.ic.propertyToProxy(\"automata.%s.Proxy\");\n"+
"		if(not %s):\n"+ /*listInterfacesIterator->getName()*/
"			raise Exception(\"could not create proxy with"%s"\")\n"+
"		self.%sPrx = %sPrx.checkedCast(%s)\n"+
"		if(not self.%sPrx):\n"+
"			raise Exception(\"invalid proxy automata.%s.Proxy\")\n"+
"		print \"%s connected\"\n"+
"\n",
		iceName,
		iceName, iceInterface,
		iceName,
			iceName, /*raise1*/
		iceName, iceInterface, iceName,
		iceName,
			iceName, /*raise2*/
		iceName
		);
	}
	printf(
"\n"
	);
}
```

Take into count that printf prints to std::cout.
For redirect to a file you can use:
* fprintf
* sprintf

PS: for example 1, you can even use '\' to avoid '+'

reysam93, 2015-11-03:
Thanks @varhub 
I haven't actually think in that. I've been doing it in that way because it how it was done in previous versions of visualHFSM, and I wanted to conserve a similar stile, but I will consider to change it.However, I don't really like using so many arguments for printf in the second example.

varhub, 2015-11-03:
Hello @reysam93,
Each printf's argument maps to each %s (because all are strings).
Advantages are only compact plain code format with embedded \t, so it can *even* be validated in code.
You can also use boost::format, which is compatible with iostreams (<<).

----
I said this approach because write workflow is little different:
1. Write code directly in Python
2. Test it?
3. Replace dynamic variables and values with %s, %d, etc
4. Paste into .cpp
5. Wrap with printf/format functions and append variables
C++ is not the best language to do it, but at least it sound a little bit clear.

PS: if boost:format allows 1:N mapping, this is what you are looking for.

If you are interested, let me known just to test boost.

varhub, 2015-11-03:
Wait was not possible!

This is the boost based code. To much pretty!
```
	cout<<format(
"       # Contact to %1%\n\
       %1% = self.ic.propertyToProxy('automata.%2%.Proxy')\n\
       if(not %1%):\n\
           raise Exception('could not create proxy with %1%')\n\
       self.%1%Prx = %2%Prx.checkedCast(%1%)\n\
       if(not self.%1%Prx):\n\
           raise Exception('invalid proxy automata.%2%.Proxy')\n\
       print '%1% connected'\n\
\n"
	) % iceName % iceInterface;

```
*Full code at https://gist.github.com/varhub/c002b6405db170eaee7c*
*g++ generate_code.cpp && ./a.out* ;) ;)

Please, revise if **%1%** and **%2%** are correct, because I'm not sure.
I also saw a possible minor bug at lines:
* https://github.com/reysam93/JdeRobot/blob/0bf2194392c941c91e72c822ae5708bd29cd9468/src/stable/components/visualHFSM/generate.cpp#L1016
* https://github.com/reysam93/JdeRobot/blob/0bf2194392c941c91e72c822ae5708bd29cd9468/src/stable/components/visualHFSM/generate.cpp#L1026

May both print the same?

reysam93, 2015-11-03:
Yes, %1% and %2% are fine, there are name  we give for the ice component, and the interface. 

And no, those two lines doesn't write the same. The first one is the first step for getting the proxy, and the second raise an exception if we haven't been able of creating the proxy.

Finally, it is the first time I heard of boost/format.hpp. How do you call it giving it different values for iceName and iceInterface. Also you have mentioned that it could create collisions. How?

varhub, 2015-11-04:
If you have never use **using namespace**, its difficult to explain it.
By the way, if both defines same funcion, what are calling? Is like
compiler error because a function has been declared twice, but with silent
behavior.

About lines xx16 and 26, Im asking if both must be getInterface(), I
supposed that, but in your code one is getName(), so Im comfused.

Lastly, if you wish (and JoseMaria let me do it) , I can do a pull request
so you can get a *final code* example.

---
Written from email.

reysam93, 2015-11-04:
Yes, you where right about line xx26, it should be listInterfacesIterator->getInterface() instead of getName, sorry.

And about the example, I would like to see it for understand completely  what you where saying about  boost, but I prefer the code of generate as it is, because I understand that completely and it doesn't have collision problems, even if the code is a bit larger.

Maybe with the example I will reconsider it.

varhub, 2015-11-04:
Collision is due `using namespace X`
Suppose:
* std::vector
* cv::vector

Then, if you use `using namespace std; using namespace cv;`, you can use directly *vector*, but.. what definition? This was I meant with collision.

This is because here I push cout<<format, but in "real" code, I change it to std::cout<<boost::format, because is more informative and will not have this problem.

----
I will do a pull request, and there you will understand how it works (I hope!).
PS: cmake correction under the hook.

jmplaza, 2016-03-09:
Solved in #333

closed 2016-03-09 by jmplaza