JdeRobot/base #216: plugin quadrotor for Gazebo: memory leak?

jmplaza, 2015-11-04:
The memory required when an application runs a simulated quadrotor continually grows, so it can not be flying long time. It seems there is a memory leak. It could be at Gazebo or (more likely) at our plugin.

fqez, 2015-11-04:
Check if the gazebo log file is growing. I had a similar issue with the nao plugins, and it was due to the log files, it grew with a lot of warning messages. The log file is <i>default.log</i> and it's placed in the gazebo directory (~/.gazebo). I will test it too.

varhub, 2015-11-10:
Here seems to be a debug code to check memory issues:
```
> togglecameraplugin.cc

void ToggleCamPlugin::OnNewFrame(const unsigned char *_image, 
		unsigned int _width, unsigned int _height, unsigned int _depth, 
		const std::string &_format) {

	if(count==0){
		image.create(_height, _width, CV_8UC3);
		count++;
	}
```

Also, `cv::Mat image` is never used.

All related logic to image storage and treatment is located/moved into quadrotorplugin
```
> quadrotorplugin.cc
QuadrotorPlugin::ImageCallback()
```

jmplaza, 2016-03-09:
Solved with solution to #229

closed 2016-03-09 by jmplaza