JdeRobot/base #217: plugin quadrotor for Gazebo: vertical bias

jmplaza, 2015-11-04:
The drone goes slowly upwards and it should not.

varhub, 2015-11-10:
Problem should be located at:
```
QuadrotorPlugin::OnUpdate()
    (navi_state == Flying) ---> link->AddRelativeForce(force);
```
And force.z must be 0 or counter gravity. Related to:
```
    force.z = mass * (controllers_.velocity_z.update(
        velocity_command_->linearZ,
        velocity.z, acceleration.z, dt) +
        load_factor * gravity);
```

Paranoid check: ensure that `velocity.z` and `acceleration.z` are **zero** when *stop*

varhub, 2015-12-18:
"Fixed" it after tune velocityZ PID parameters.
This was issue before and after.
Accurate fix (takeoff) requires **fix_position** commands (go to XYZ)

jmplaza, 2016-03-09:
Solved with solution to #229

closed 2016-03-09 by jmplaza