JdeRobot/base #220: Missing the configuration files (.cfg) of turtlebot plugin with the new debian packages

almartinflorido, 2015-11-08:
Hi all,

yesterday I installed on a virtual machine (x64) the latests debian packages and the configuration files of turtlebot (cam_sensor_left.cfg, turtlebot_laser.cfg, etc.) plugins doesn't appear in the default directory (/usr/local/share/jderobot/gazebo/plugins/turtlebot).

regards

fqez, 2015-11-08:
Hi Alberto, they are all in the /usr/local/share/jderobot/conf

almartinflorido, 2015-11-08:
Ok, Francisco but for pioneer/flyingball/kinect/nao/quadrotor plugin the files they are in /usr/local/share/jderobot/gazebo/plugins/<plugin>, no? 
Maybe I copied this files and I don't remember.

thanks

fqez, 2015-11-08:
Yes, you are right. I modified the turtlebot plugin some time ago, and I decided not to copy the .cfg files again, I think it makes no sense to have them duplicated. I will tell JoseMaria if we can remove .cfg files of the others plugins directories, or we keep the .cfg files also in the plugin folder instead.

almartinflorido, 2015-11-08:
Ok @fqez  thanks

closed 2015-11-10 by fqez