JdeRobot/base #221: The kinect mesh don't appear in turtlebot model

almartinflorido, 2015-11-08:
Hi all,

in ardrone-turtlebot.world the path to kinect mesh is commented and it is incorrect, the result is that the kinect mesh don't appear in the turtlebot model
[image]

The correct path is //kinect/meshes/kinect.dae instead of of //kinectJde/meshes/kinect.dae

[image]

regards

fqez, 2015-11-08:
Hi Alberto, thank you for reporting this. The current world ardrone-turtlebot is not correct. It redefines the existent turtlebot model (which should be at ~/.gazebo/models. I'm goingo to update the world to solve this issue. 

Regards!

almartinflorido, 2015-11-08:
Thanks @fqez

fqez, 2015-11-08:
You can test it now, if you want! pull request #226

almartinflorido, 2015-11-08:
Thans @fqez  maybe later I can test it

commit 3e6678ea (2015-11-11): Merge pull request #226 from fqez/issue_221

Fix #221 Updated the ardrone-turtlebot.world and solved the mesh issue
commit 54d86100 (2015-11-11): Fix #221 Updated the ardrone-turtlebot.world and solved the mesh issue
closed 2015-11-11 by fqez