JdeRobot/base #222: The car model of car_apritalgs.world show white wheels in virtualized environment

almartinflorido, 2015-11-08:
Hi all,

yesterdary I installed a virtual machine (x64) with the new debian packages, I tested the car_apriltags.world with introrob_py and the images received by the plugin show white the wheels car.

[image]

Does anyone know that happens?

regards

fqez, 2015-11-08:
Hi @almartinflorido 

the problem is in these lines of the <i>model.sdf</i>:

```
<collision name="collision">
    <geometry>
        <cylinder>
            <radius>0.05</radius>
            <length>0.1</length>
        </cylinder>
    </geometry>
</collision>
<visual name="visual">
    <geometry>
        <cylinder>
            <radius>0.05</radius>
            <length>0.1</length>
         </cylinder>
     </geometry>
</visual>
```

int these links:

```
<link name="{rear,front}_{left,right}_steering_wheel_block">
<link name="{rear,front}_{left,right}_wheel">
```
As long as they only specify visual properties, they are not required. If you comment out this block of code for each wheel in (<i>~/models/carApriltag/model.sdf</i>), you can remove this visual effect. 

I am not sure, but maybe gazebo primitive shapes have visual preferences over textures.

Try it and let me know if you success!

Regards!

almartinflorido, 2015-11-08:
Yes, with this changes the car don't show the white wheels.

Thank you so much @fqez !!

commit bddad6ee (2015-11-10): Fix #222 removed wheels visual properties of carApriltag model
commit f9c62186 (2015-11-10): Merge pull request #228 from fqez/issue_222

Fix #222 removed wheels visual properties of carApriltag model
commit d53920c0 (2015-11-11): Merge pull request #232 from fqez/issue_222

Fix #222 Complete pull request #228
closed 2015-11-10 by fqez