JdeRobot/base #223: The ultrasonic altimeter of ardrone plugin doesn't works

almartinflorido, 2015-11-08:
Hi all,

the altimeter of ardrone plugin always return 0, the next image show a drone over a car (aprox two meters in height) for this case the altimeter return 0 and the z axis of absolute pose 3D 4.65m.
[image]

regards

fqez, 2015-11-08:
Hi @almartinflorido 

the problem is in these lines of the <i>model.sdf</i>:

```
<collision name="collision">
    <geometry>
        <cylinder>
            <radius>0.05</radius>
            <length>0.1</length>
        </cylinder>
    </geometry>
</collision>
<visual name="visual">
    <geometry>
        <cylinder>
            <radius>0.05</radius>
            <length>0.1</length>
         </cylinder>
     </geometry>
</visual>
```
As long as they only specify visual properties, they are not required. If you comment out this block of code for each wheel in (<i>~/models/carApriltag/model.sdf</i>), you can remove this visual effect. 

I am not sure, but maybe gazebo primitive shapes have visual preferences over textures.

Try it and let me know if you success!

Regards!

fqez, 2015-11-08:
Maybe related with this issue #196?

almartinflorido, 2015-11-09:
No, the functionality has never been implemented I think.

varhub, 2015-11-09:
I'm reviewing quadrotor plugin at now, and if altimeter info is part of navdata struct, it is not implemented.

I suppose that it would require something like raytracing.

varhub, 2015-12-15:
Added to #229 changeset

closed 2016-01-19 by fqez