JdeRobot/base #224: Discussion: port table for JdeRobot interfaces/components

almartinflorido, 2015-11-08:
Hi all,

I want to create a table with the default ports used in JdeRobot. The idea is to have a table in JdeRobot's wiki with all ports used. I propose two approach:

- A table with interface -> default port, for example:

| Interface        |Port           | 
| ------------- |:-------------:| 
| camera      | 9999 |
| motors      | 8554      | 
...

- A table with component -> interface -> default port, for example:

| Component               | Interface  | Port    |
| --------------------------- |:-------------:| --------:|
| basic_component     | camera     |  9999  |
| ardrone_server         | camera     |  8999  |
|                                  | navdata     | 8998   |
...

any suggestions?

regards

fqez, 2015-11-08:
Yes, that's exactly what we need right now. I am modifying the jderobot manual, and it would be quite useful to have a port convention. At the moment, we have to look for the ports of all plugins we want to use in the terminal in order to configure the .cfg files porperly, it's a waste of time!

varhub, 2015-11-09:
About this, I opened #199 some time ago.
For "components" or robots, single port is simple and better.
Perhaps, actuators and cameras should also have its own port.
By the way, if all robots' cameras are targeted to same port, it is not possible to have more than one.
This will enforce sequential testing.

We should do a core library for register "common" interfaces.
Ports can be hardcored, and it can be enabled or diabled via arguments.

This core library will act as NOOP when disabled, but will provide standard interfaces through standard ports when enabled.

varhub, 2015-11-19:
About that library, I think that better name will be:
*Global.A.a=B*
Example:
*Global.Camera.Endpoint=...*

closed 2021-02-21 by jmplaza