JdeRobot/base #225: Discussion: world and configuration files names

almartinflorido, 2015-11-08:
Hi all,

I want to purpose a style guide for world and configuration files. Actually the worlds of gazebo they are as follows.

| Actually                          | Proposed                         | Notes  |
| --------------------------------|:---------------------------------:| ---------------:|
| ardrone-beacons            | adrone-beacons              | The name describes the world |
| ArDrone_boxes              | ardrone-boxes                 | For naming style |
| ArDrone_simple-gsoc    | ardrone-gsoc-simple        |  A world without textures   |
| ArDrones                        | ardrones-simple               | Two drones in a simple world   |
| ArDrones_texture_gsoc | ardrone-gsoc-texture        |    |
| ardrone-turtlebot            | ardrone-turtlebot-simple    | A world with ardrone and turtlebot    |
| ArDrone                         | ardrone-simple                   | One drone in a simple world |
| car_apriltag                    | ardrone-car-apriltags         | One drone and one car in a world with apriltags |
| departamental2              | pioneer-departamental      | One pioneer in departamental world |
| fbworld                           | flyingball-simple                 | |
| gato_raton_1                 | ardrones-gato-raton           | A world with two drones for gato ratón  |
| gato_raton_2                 | -                                          | Equal than gato_raton_1 with drones in differents positions|
| jaramaSigueLineas        | pioneer-jarama-siguelineas |  |
| kinect                             | kinect-simple                        |  |
| montmelo-kobuki           | kobuki-montmelo                 |   kobuki or turtlebot? |
| montmelo-pioneer          | pioneer-montmelo               |  |
| naoworld                        | nao-simple                           |  |
| pioneer2dxJde-2            | pioneer-simple                     |  |
| pioneer2dxJde               | pioneer-labyrinth                  |  |
| plane_demo                   | plane-simple                        |  |
| quadrotor                       | quadrotor-simple                  |  |
| reconstruccion3D          | pioneer-3dreconstruction     |  |
| road_drone                    | ardrone-road-simple             |  |
| road_drone_textires       | ardrone-road-textures          |  |

The general idea is:
```
<actor or actors>-<world description>-<action description>.world
```

For configuration files the idea is:
```
<actor>-<interface name>-<description>.cfg
```
Example:
turtlebot-camera-rigth.cfg

any suggestions?

regards

fqez, 2015-11-08:
I agree whit this. We better have a naming convention in order to prevent future headaches.

varhub, 2015-11-09:
Mmm... it is a very valid proposal.
By the way, just some questions
1. What is most important, robot or scene?
2. Who could be imported as "include"?
3. Plain world files or include based?

Naming sounds good, but requires '_' for spaces.
For example:
ardrones-gato-raton will be ardrones-?-gato_raton

Since not every world have a clear context (gsoc, curso drones, test, etc) something like:
context-actors-world-actions seems unpractical

My second mini change is replace plurals by _xN

Finally, if world is more important, names could be:
empty-ardrone_x2-gato_raton
departamental-pioneer
empty-ardone_car-follow_apriltag

First word is really important for indexing, so i suppose that will be difficult reach a clear naming convention

closed 2021-02-21 by jmplaza