JdeRobot/base #227: Some worlds doesn't works in gazebo

almartinflorido, 2015-11-08:
Hi all,

I tested some worlds than doesn't works in gazebo the worlds are:

-Ardrones.world, only appears one drone.
-jaramaSiguelineas.world
-montmelo-kobuki.world
-naoworld.world

can anyone confirm it?

regards

fqez, 2015-11-08:
Hi!

* ArDrones.world does not work because a model is missing: ArDrone_Stick. It does not exist.
* jaramaSiguelineas.world does not even load. It seems it has something to do with <i>jarama</i> model
* montmelo-kobuki.world same as jaramaSiguelineas.world
* naoworld.world I'have this world fixed from my PFC, but I will upload all the nao updates when I have some time, so this one is not that important.

Thanks!

closed 2016-03-09 by fqez

fqez, 2016-03-09:
Solved at #370 
ArDrones.world is deprecated and will be erased.
jaramaSigueLineas fixed in #371
montmelo-kobuki fixed in #370 
naoworld is not availabe yet because the nao support is going to be revisited.

commit 5b2af4d4 (2016-03-09): [issue #227] Fixed jaramaSiguelineas.world. Fix #227
commit 8dc14451 (2016-03-09): Merge pull request #371 from fqez/issue_227

[issue #227] Fixed jaramaSiguelineas.world. Fix #227