JdeRobot/base #233: How to set the size of AprilTag in the car model?

weiweikong, 2015-11-11:
In order to get the distance between the Ardrone and Car, the real length of the Apriltag should be known. How could we set the size correctly ?

I am personally not familiar with the model construction, so I only checked `models/carApriltag/model.sdf`and `models/carApriltag/meshes/model.dae`files, but could not find any detail about the size. Is the size related to the `png`file ?

weiweikong, 2015-11-13:
@almartinflorido @fqez do you have any experience about this ?

fqez, 2015-11-13:
Hi @weiweikong,

sorry for the late response. As far as I know the car model doesn't allow to set the size of the beacons or tags. I guess we do the tracking of the tags using the AprilTags libraries, but @almartinflorido knows how it works better than me. The sure thing is that the current model doesn't be modified in any way. 
I will imporve it to make it more configurable.

weiweikong, 2015-11-13:
@fqez As far as I know, I print out Apriltag and set the real length to get the distance and rotation between the tag and camrea. Therfore the measure of distance from Apriltag is based on the true length of the tag and the measure of rotation is depend on the code of each ID. So without the true length, we could not know the true distance between the car and camera in Gazebo. Is that right ?

fqez, 2015-11-13:
Yes, that's the the theory, but I don't know how @almartinflorido do it in Gazebo. I will tell him to answer you if he has some time!

weiweikong, 2015-11-13:
Great! Thank you for your reply!~ @fqez

almartinflorido, 2015-11-14:
Hi @weiweikong,

unfortunately I have not used apiltags in my projects but I think that the world is not configurable, you maybe can change the tag but didn't specify a new size. Anyway I sure that @andresjhe knows the size of the tag.

If you not have the size you can obtain the relative position of the camera.

regards

jmplaza, 2015-11-14:
@weiweikong you can also take a look at the work of Alberto López-Cerón (http://www.jderobot.org/Alopezceron-tfm) who worked with Apriltags in Gazebo and distance estimation of the camera to the visual tags. You can also access to their Gazebo worlds in his svn repository (https://svn.jderobot.org/users/alopezceron/tfm/trunk/)

weiweikong, 2015-11-14:
@jmplaza Thank you for your reply. I will check that. :)

andresjhe, 2015-11-14:
Hi @weiweikong! Sorry, I was busy and I couldn't answer you before. 
If I'm not mistaken, you can know the distance if you have the focal lenght of the camera you are using (in this the ArDrone simulated camera). The work of Alberto López-Cerón has more info about it.
If you still want to change the size of the AprilTag I suggest you do one of the following options:
- Scale the Gazebo model. There's a property in the .sdf files where you can set the size you want.
- I can provide you with a car model without the tag, and you can add on top a rectangular shape with your desired size and use the Apriltag as material. Then, join them together in Gazebo. 
- I can provide you the blender file where you can extract the .dae and make the Apriltag smaller.

The second option sounds harder then what it is, but it was faster for me to just apply the Apriltag in the .dae model in Blender. I hope it helped you!

alopezceron, 2015-11-14:
Hello @weiweikong,

I am not sure of how you are using the Apriltags, but, as @jmplaza has pointed out, I worked including them in the gazebo .world file, the following way:

\<size\>1.000000 1.000000 0.100000\</size\>, which means: 1m height, 1 m width and 10 cm thickness.

You can find an example here: https://svn.jderobot.org/users/alopezceron/tfm/trunk/gazebo/ArDrone_AprilTags.world. I think you need to set the size in both "collision" and "visual" xml nodes.

Hope that helps :-)

weiweikong, 2015-11-15:
Hi @andresjhe and @alopezceron , thank you for your informative reply.
In my experience, I tried to add the Apriltag on a Husky mobile robot cooperate with an ardrone which running pixhawk modules in Gazebo, as shown in following figure.
[image]
However, 
- the Apriltag seems to published as a texture which extend on the surface, although my Apriltag file only contain one tag.

[image]
- Also, I could not get the real size of the tag in Gazebo.

I just modified `top_plate_texture.dae` which called from `decorations.urdf.xacro`
```
...
<link name="top_plate_texture_link">
        <visual>
          <origin xyz="0 0 0" rpy="0 0 0" />
          <geometry>
            <mesh filename="package://husky_description/meshes/top_plate_texture.dae" />
          </geometry>
        </visual>
 </link>
...
```
and in `top_plate_texture.dae` file
```
...
<library_images>
    <image id="plywood256x256_jpg" name="plywood256x256_jpg">
      <init_from>apriltag-36h11-0.png</init_from>
    </image>
 </library_images>
...
```
You can check all the files at https://github.com/CLDrone/rotors_simulator/tree/husky/husky_gazebo/urdf
It seems no paramters related to the pyisical size of the `top_plate_texture`. Also, I did not expect the Gazebo extend my tag when the it was smaller than the surface.

As you suggestion, I checked the `gazebo/models/carApriltag/meshes/model.dae`. There is similar configuration
```
...
 <library_images>
    <image id="tag36_11_00008_png" name="tag36_11_00008_png">
      <init_from>tag36_11_00008.png</init_from>
    </image>
 </library_images>
...
```
Also, there is no specific paramters to set the tag size, but as on the car surface, it was only one texture not the same as mine, which has extend the marker. How could I set the size of the top rectangluar shape as you suggested correctly ?

varhub, 2015-11-15:
@weiweikong, perhaps texture problem is due to "repeat" settings into texture.
You can try to enlarge geometry to see is it is the same apriltag.

weiweikong, 2015-11-15:
@varhub Hi, it was possible to enlarge the tag, however, it was not possible to get the real size of Apriltag. Therefore the true distance between the camera and tag can not be achieved.

jmplaza, 2015-11-15:
This is not a JdeRobot question itself but @weiweikong maybe you can find the tag size in a 'reverse engineering' way: assume certain size (s'), use your 3D localization method to estimate the distance between the camera and the tag, then get from Gazebo the true 3D localization and compare to the estimated one. Modify s' until estimated and true distances (simulated) match. Hope it helps.

weiweikong, 2015-11-17:
@jmplaza Yes, it is true that the basic configuration is relate to Gazebo environment. I'd like to test as you suggested.

closed 2016-06-22 by fqez