JdeRobot/base #234: Discussion: Reorganization of code

fqez, 2015-11-12:
Hi, 

we should be able to reorganize our code in a better way than components and drivers mixed all together. In a meeting with JoseMaria, we agreed to reorganize the project following this schema:

```
Jderobot
└─ src/stable
    ├─ tools
    │   └─ basic_component
    │   ├─ cameraview
    │   ├─ introrob
    │   ├─ ...
    │ 
    └─ drivers
    │   └─ basic_server
    │   ├─ cameraserver
    │   ├─ gazeboserver
    │   ├─ ...
    └─ miscellaneous or others (to uncategorized tools)
    └─ interfaces (no changes)
    └─ libs (no changes)
```

What do you think?

weiweikong, 2015-11-13:
It should be better for the beginners.

varhub, 2015-11-13:
Instead tools, could be possible to call it frontend?
Or a name that implies that is mainly the client version (or "a" client), with also GUI

Perhaps tools word should be reserved for cmake, install or other stuff.

fqez, 2015-11-13:
Do you think that name is intuitive for the beginners? I guess it should be easy to understand for the new people, but it's just my opinion. What the rest of you guys think?

Thanks for contribute. :)

jmplaza, 2015-11-13:
For me it is fine. It is a better distribution than the current generic "components" directory. The idea is to group into the Drivers directory the components which provides access to real (or simulated) sensors and actuators.

Most of the remaining pieces of software are tools of the platform which provide a variety of functionality: camera calibrator, recorder+replayer, visualHFSM tool, example component (basic_component) and many viewers and teleoperators (cameraviewer, rgbdviewer...). 

The proposed division introduces more semantic into the collection of software pieces. A little step in a good direction, without adding thousand directories more.

fqez, 2015-11-13:
So we use "tools" or "frontend" finally?

varhub, 2015-11-14:
Tools.
because frontend do not cover all stuff.
fronend, or even clients, will be valid for basic_component, introrob, cameraview, teleoperator , ...
But teleoperator is a tool? A client? Where place it?

By now only tools could be ok, but there is a further question:
Can cameraview be used as library? So there is not needed to recode ice ?
If true, where place it?

As summary
@jmplaza wants to avoid a flatten directory, so only split into drivers and tools

varhub, 2015-12-11:
@fqez, reopen issue to track it.
I think that a minor step can be done just for visualHFSM, replayer, etc in a future but near release.
(Might this change increase patch or even minor due it "breaks" paths and even worse, git history?)

varhub, 2015-12-17:
After see http://jderobot.org/index.php/Tools I think that a valid word will be **managers**.
Why? Because managers are a specific client side of driver with a specific behavior. They just provide a way to interact with a concrete driver.

PS: what should be a **tool**? anything generic do not depends on concrete driver: CameraCalibration, Replayer, and so  on.

*A generic manager is a tool because you create a driver compatible with it, but not reverse.*

This means that by creating simple managers (as much generic as interfaces) you can eventually use it and compose to make an ad-hoc manager. Just to avoid write and rewrite exactly same code.
This belong Qt and Widget based interfaces will provide a block based component composition (for me: "pinta y colorea")

varhub, 2015-12-22:
**Red flag!** by do this change, install sentences at root CMakeLists will no longer work. We will need to pay attention here.

jmplaza, 2015-12-26:
As part of the reorganization, those components which are related to teaching (introrob, introrob_py, introrob_qt...) will be moved to a different repository JdeRobot-TeachingRobotics

fqez, 2016-06-22:
This is already done.

commit b82f5341 (2015-11-23): Updated CMakeLists.txt with the new organization of the project.

Fix #234
commit ff669f47 (2015-11-23): Merge pull request #242 from fqez/issue_234

Fix #234 reorganization of project code
closed 2016-06-22 by fqez