JdeRobot/base #235: jderobot package: gazebo error when try to run plugins

lr-morales, 2015-11-13:
Hello all,

I have done a update to the current version of the jderobot (13-nov-2015) packages from the previous one (installed at 9-nov-2015) and after copying the models and world as described at the _Advanced drone programming course_ , when running gzserver I got this output:

    Gazebo multi-robot simulator, version 5.1.0
    Copyright (C) 2012-2015 Open Source Robotics Foundation.
    Released under the Apache 2 License.
    http://gazebosim.org
    
    [Msg] Waiting for master.
    [Msg] Connected to gazebo master @ http://127.0.0.1:11345`
    [Msg] Publicized address: 10.1.36.17
    [Err] [Plugin.hh:153] Failed to load plugin libtogglecamplugin.so: libtogglecamplugin.so: cannot open shared object file: No such file or directory
    [Err] [Plugin.hh:153] Failed to load plugin libtogglecamplugin.so: libtogglecamplugin.so: cannot open shared object file: No such file or directory
    [Err] [Plugin.hh:153] Failed to load plugin libimuplugin.so: libimuplugin.so: cannot open shared object file: No such file or directory
    [Err] [Plugin.hh:153] Failed to load plugin libquadrotorplugin.so: libquadrotorplugin.so: cannot open shared object file: No such file or directory
    [Err] [Plugin.hh:153] Failed to load plugin libposeplugin.so: libposeplugin.so: cannot open shared object file: No such file or directory
    [Err] [Plugin.hh:153] Failed to load plugin libcarMotors.so: libcarMotors.so: cannot open shared object file: No such file or directory

After checking some possibilities, I found that the dynamic libraries named in the output didn't have execution permissions. Running a `chmod ugo+x` over those plugins solved the issue, but gazebo should be able to run those when doing a fresh jderobot package install.

Could someone confirm if this is a general issue?

This system is a fresh xubuntu 14.04.3 (LTS) installation.

Regards,
    L. Roberto Morales Iglesias

fqez, 2015-11-13:
Hi @lr-morales,

which package did you install? and which architecture are you working on? I tried both packages in Ubuntu 14.04 32-64bits and they work fine. 
Also, can you check please if those plugins are in the correct directory (/usr/local/share/jderobot/gazebo/plugins/)?
You should source the gazebo-setup.sh before start gazebo too.

Regards!

lr-morales, 2015-11-13:
Hello,

I installed both jderobot-core and jderobot-deps and I'm using an x64
architecture for this.
The plugins are in the correct directory (I had to find them to perform the
chmod, but re-checked to be sure) and gazebo-setup.sh is sourced correctly.

Regards.

fqez, 2015-11-13:
Ok I will wait for someone to report the same problem, cause @almartinflorido tested both packages too and it worked too. 

Thanks for the report!

varhub, 2015-11-14:
+1
Installed from sources and permisions are 644

cmake automatically set 755 permisions for executables, but i dont known for libraries.

By the way, check if these libraries are installed like files or programs (related to issue #)

almartinflorido, 2015-11-14:
Hi all,

I didn't had any problems with the installation or gazebo. I think that @lr-morales and @varhub uninstall a previous version of JdeRobot and after install the new version,  right?.

My configuration is a x64 bits with Ubuntu 14.04 without previous versions of JdeRobot.

regards

varhub, 2015-11-14:
-1

My error seems to be due outsource build
`[Err] [Plugin.hh:153] Failed to load plugin /usr/local/share/jderobot/gazebo/plugins/quadrotor/libquadrotorplugin.so: libcolorspacesmm.so:`
Because `libcolorspacesmm.so` was not been installed.

By the way, I checked my development chroot, whose do not complain about -x permission.
```
varribas@Ubuntu14lts:/home/datos/chroot/Hosted$ find JdeRobot5-git/usr/local/ -path '*gazebo/plugins/*' -name '*.so' -ls
2838173  396 -rw-r--r--   1 root     root       404026 sep  2 21:32 JdeRobot5-git/usr/local/share/jderobot/gazebo/plugins/turtlebot/liblaserTurtlebotJde.so
2834311 1128 -rw-r--r--   1 root     root      1152436 nov 12 17:04 JdeRobot5-git/usr/local/share/jderobot/gazebo/plugins/quadrotor/libquadrotorplugin.so
2834312  356 -rw-r--r--   1 root     root       362709 nov 12 15:13 JdeRobot5-git/usr/local/share/jderobot/gazebo/plugins/quadrotor/libposeplugin.so
```
So libraries do not requires exec permission nor gazebo has been updated and now requires it*.

Both have same Gazebo version `gazebo5  5.1.0-1~trusty`

---
@lr-morales , check if .so directories are listed into plugin path (`export -p | grep GAZEBO`).
What version have you got of gazebo?

lr-morales, 2015-11-20:
Hello there,
 I have repeated the installation process (following again the instructions from the drone course) in a new laptop and it seems to work correctly now. The only he between both installations (excluding the laptop) is that in the first case I uptated from the previous version.

gazebo version is now 5.1.0
the `export -p | grep GAZEBO` contains the plugin directories inside the correct variable.

Thanks for your help, and if I notice more differences or possible causes, I'll post them here.

varhub, 2015-12-11:
closeable

closed 2015-12-12 by fqez